e2nIEE/pandapower #340: FutureWarning when performing runpp with init_va_degree

dlohmeier, 2019-03-19:
I encountered a warning when performing a loadflow with values to initialize vm_pu and va_degree. This doesn't seem to harm the calculation itself, as the return value of the if-statement shown below will probably always be False for the given case. However the internal conversion of options in `runpp` seems to lead to this error, which could probably be dealt with:

/home/dbuechner/pandapower/pandapower/pf/run_newton_raphson_pf.py:44: FutureWarning: elementwise comparison failed; returning scalar instead, but in the future will perform elementwise comparison
  if options["init_va_degree"] == "dc":

So especially if in future this comparison will return a full numpy array or pandas series, it will then cause an exception. 
A simple adaptation to ask for the type of the option should be sufficient, or does someone disagree with that?

if isinstance(options["init_va_degree"], str) and options["init_va_degree"] == == "dc":
     ... perform dc loadflow here

lthurner, 2019-03-20:
Yes good point, please submit the change in a PR

lthurner, 2019-03-20:
Closed in #341

closed 2019-03-20 by lthurner