e2nIEE/pandapower #348: determine stubs returns wrong stub bus

FlorianShepherd, 2019-03-29:
I have  a simple grid, which looks like this:

            b0
            |
            trafo
            |
        b6--b1--b2
        |   |   |
        b5--b4--b3

The function `topology.determine_stubs` returns that b6 is a stub bus and the connected lines are on a stub. This seems to be a bug in `find_graph_characteristics` 

@jkupka could you have a look at this? I'll add a test

FlorianShepherd, 2019-03-29:
> I have a simple grid, which looks like this:
>
> ```
>         b0
>         |
>         trafo
>         |
>     b6--b1--b2
>     |   |   |
>     b5--b4--b3
> ```
>
> The function `topology.determine_stubs` returns that [...]

Here is the test:
https://github.com/FlorianShepherd/pandapower/commit/db18a2fcb8c50d92a4390e527921804c554dc45c

jkupka, 2019-04-01:
I'm not able to reproduce the result you described with the network you're using for the test.

For me `determine_stubs` returns that all buses except b0 are stub buses, which is expected behaviour. There is only one connection from b0 to b1 so all buses after this connection are determined as stub buses.
If i use b1 as root bus (`determine_stubs(meshed_network, roots=[1]`) b1-b6 are not determined as stubs anymore. 

If no root buses are declared, `determine_stubs` will use the connection buses of external grids per default. One could question if this is the proper approach.

jkupka, 2019-04-04:
I was able to reproduce the wrong behaviour and fixed it with #353. In some cases the order of the bus discovery during the graph search led to falsely determined stub buses.

PR #353 (2019-04-05): fixed identification of stub buses for some egde cases and added a test
fix #348
closed 2019-04-05 by jkupka