e2nIEE/pandapower #350: First time running - TypeError "p_mw" issue

dogeraldi05, 2019-04-02:
Hello, 
Somehow the simple power flow (example below) show the issue:

_" TypeError: create_load() got an unexpected keyword argument 'p_mw' '_

The code below:
**---**
import pandapower as pp
#create empty net
net = pp.create_empty_network() 

#create buses
b1 = pp.create_bus(net, vn_kv=20., name="Bus 1")
b2 = pp.create_bus(net, vn_kv=0.4, name="Bus 2")
b3 = pp.create_bus(net, vn_kv=0.4, name="Bus 3")

#create bus elements
pp.create_ext_grid(net, bus=b1, vm_pu=1.02, name="Grid Connection")

pp.create_load(net, b3, p_mw=0.1, q_mvar=0.05, name="Load")

#create branch elements
tid = pp.create_transformer(net, hv_bus=b1, lv_bus=b2, std_type="0.4 MVA 20/0.4 kV", name="Trafo")
pp.create_line(net, from_bus=b2, to_bus=b3, length_km=0.1, name="Line",std_type="NAYY 4x50 SE")
**----**

How can I fix this "TypeError " issue ?

lthurner, 2019-04-02:
You probably have an old version of pandapower, run
```
pip install pandapower --upgrade
```
and try again.

dogeraldi05, 2019-04-02:
It works,  thanks a lot @lthurner !

closed 2019-04-02 by dogeraldi05