e2nIEE/pandapower #351: Tracing upstream and downstream

GirishVC86, 2019-04-03:
Are there any functions available for tracing upstream to source or downstream trace from source to sink.

lthurner, 2019-04-03:
I think you will have to describe more precisly what you mean by that?

GirishVC86, 2019-04-10:
Tracing is used to find all features connected to the same feeder system. Upstream trace from Load to Source, downstream trace from Source to all connected Load with intermediate elements. If I change the switch position within given network then Line element parameter "From" and "To" does not change.

lthurner, 2019-04-10:
Check out the topology module, I think it does exactly what you are looking for: https://github.com/e2nIEE/pandapower/blob/develop/tutorials/topology.ipynb

lthurner, 2019-05-07:
Closed for inactivity, if you have any more questions feel free to reopen.

closed 2019-05-07 by lthurner