e2nIEE/pandapower #357: topology - Why does unsupplied buses have a parameter in_service only?

julffers, 2019-04-09:
The topology function unsupplied buses has a parameter "in_service_only". If True, unsupplied buses, that are out of service are removed from the returned result. Why does this paramter exist? If a pandapower net and no multigraph are passed to unsupplied buses, create_nx_graph creates a multigraph that does not contain out of service buses anyway.

Is there any reasonable case, where one would pass a net AND a multigraph that does not correctly represent the net and therefore, the out of service buses would have to be removed from the result?

lthurner, 2019-04-10:
I remember @rbolgaryn requested this feature, but looking at it now I am also not sure why it exists. The implementation is also not very efficient with iterating over the buses and then indexing with .loc. Seems like a functionality that can be deprecated to me.

FlorianShepherd, 2019-04-15:
Please removetthis feature @julffers

closed 2019-05-06 by lthurner