e2nIEE/pandapower #364: Negative line losses

meyer89, 2019-04-12:
I stumbled upon negative line losses results and wondered how this might happen. So I looked up the code and found the following calculation of losses in the function _get_line_results().

        pl_kw = pf_kw + pt_kw
        ql_kvar = q_from_kvar + q_to_kvar

It might be rare, but negative losses are possible and this does not make any sense. A simple fix would the calculation of absolute difference between FROM and TO power values.

        pl_kw = np.abs(pf_kw + pt_kw)
        ql_kvar = np.abs(q_from_kvar + q_to_kvar)

lthurner, 2019-04-12:
> It might be rare, but negative losses are possible and this does not make any sense.

Cany you give an example with negative line losses? From my understanding it would only be possible if you defined a negative line impedance, in which case negative losses would actually be the mathematically "correct" result.

jhmenke, 2019-04-16:
can this be closed? unless the line is somehow generating energy, pl_kw will always be positive

meyer89, 2019-04-16:
I have lines in a network with negative losses but positive impedance. I did not have the time yet to extract the relevant lines to show an example.

This is one line which produces negative pl_kw:
length_km       0.000576552
r_ohm_per_km          0.248
x_ohm_per_km           0.08
c_nf_per_km             256
g_us_per_km               0
max_i_ka               0.27
df                      1
parallel                  1
type                     cs
in_service             True

One thing which might lead to the negative result is the extremly low length of the line (sometimes 1m or less). So yeah it might be a rounding problem or other inaccuracy in the calculation.

I did not think about the negative impedances, so the behavior right now might be the best solution even if it leads to negative pl_kw in some rare cases.

jhmenke, 2019-04-16:
what is the magnitude of the negative loss in your case? and what are pf_kw and pt_kw?

meyer89, 2019-04-16:
The error for pl_kw is within the range of power mismatch after solving the power flow and can therefore only happen in scenarios with very low load or generation.

A quick check convinced me that the calculation is correct as it is, even when negative "losses" are confusing at first.

net.res_ext_grid.q_kvar + net.load.q_kvar.sum() + net.sgen.q_kvar.sum() + net.storage.q_kvar.sum() + net.res_line.ql_kvar.abs().sum() + net.res_trafo.ql_kvar.sum()
>> 0.019862
net.res_ext_grid.q_kvar + net.load.q_kvar.sum() + net.sgen.q_kvar.sum() + net.storage.q_kvar.sum() + net.res_line.ql_kvar.sum() + net.res_trafo.ql_kvar.sum()
>> 6.552057e-10

closed 2019-04-16 by meyer89