e2nIEE/pandapower #374: current limited voltage source

hariramtud, 2019-04-22:
hallo developers,

Is it possible to model a "Current limited voltage source" in padapower ?

jurasofish, 2019-04-23:
You're perhaps looking for a static generator

hariramtud, 2019-04-23:
Thanks. I am not looking for a static generator. I am looking for a method to create a DC offset in the LV side of the transformer through a current limited voltage source which will create a Virtual neutral point. I want to model this to find the short circuit level at that point where the current limited voltage source is connected.

lthurner, 2019-05-06:
sgens can be modeled as a constant current source if you want to assign a fixed current to an element. I don't know how a current limited voltage source would be modeled. What you describe does not sound like anything that is within the IEC 60909 standard, which the pandapower short-circuit module implements. It is detailed [here](https://pandapower.readthedocs.io/en/v2.0.1/shortcircuit/ikss.html#voltage-source-contribution) how the short-circuit current is calculated. I would not know how to implement something like a current limited voltage source in this framework.

lthurner, 2019-06-24:
Closed for inactivity, please reopen if you want to discuss this furthere or have a specific question.

closed 2019-06-24 by lthurner