e2nIEE/pandapower #377: runopp cost optimal solution

daetzm, 2019-05-01:
Hi everyone,

When I run my OPF simulations, runopp (pandapower version 1.6) does not find the cost optimal solution. I have a generator that is around 10 times cheaper than another one and yet both produce around 50% of their capacity. In the cost optimal solution the cheaper generator should run at 100%. I checked the cost functions multiple times and also changed them in order for the cheaper generator to have 0 costs for the entire production range. This also didn't change the outcome.

Thanks for your help
Max

friederikemeier, 2019-05-02:
Hi, that is weird. Could you please provide a simple code example so I can check it?

TianmengWhu, 2019-05-05:
Hi, I also find that the gencost is negtive. But the gencost is positive in Matpower. Hence, the total cost, i.e, net.res_cost is negtive in pandapower. Why? 
Thanks for your help.
Meng

lthurner, 2019-05-06:
That is because the signing systems in pandapower 1.6 are different than in MATPOWER (generated power was negative in pp 1.6 opposed to positive in MATPOWER). In pandapower 2.0, the signing system was changed to match that of MATPOWER and other tools as documented [here](https://pandapower.readthedocs.io/en/v2.0.1/about/update.html). If you switch to pandapower 2.0, everything will have the same sign as in MATPOWER.

lthurner, 2019-05-06:
> When I run my OPF simulations, runopp (pandapower version 1.6) does not find the cost optimal solution.

A lot has been changed between pandapower 1.6 and pandapower 2.0 with regards to the OPF. The signing system of generators was changed, the way to define the cost functions was changed and several bugs where fixed. We can't support old versions of pandapower, please [update to 2.0](https://pandapower.readthedocs.io/en/v2.0.1/about/update.html) and see if the problem remains. If it doesn, post a minimal examples that reproduces the problem so that we can have a look at it.

TianmengWhu, 2019-05-06:
> When I run my OPF simulations, runopp (pandapower version 1.6) does not find the cost optimal solution.
> 
> A lot has been changed between pandapower 1.6 and pandapower 2.0 with regards to the OPF. The signing system of generators was changed, the way to define the cost functions was changed and several bugs where fixed. We can't support old versions of pandapower, please update to 2.0 and see if the problem remains. If it does, post a minimal examples that reproduces the problem so that we can have a look at it.

Actually, I use the pandapower 2.0. I find it is not the truth that the signing system was changed to match that of MATPOWER regards to the OPF.  First, the it is said that "Bus power values are given in the consumer system. Therefore a bus with positive p_mw value consumes power while a bus with negative active power supplies power"(https://pandapower.readthedocs.io/en/v2.0.1/elements/bus.html?highlight=res_bus). Second, the  total cost, i.e, net.res_cost is negtive in pandapower 2.0. 

Here is the code.

`import pandapower.networks
import pandapower as pp
net = pandapower.networks.case30()  #  import data
pp.runopp(net)  # the optimal power flows as the initial state
print(net.res_cost)`

-634.8217051637471

Thanks.

lthurner, 2019-05-06:
>  I find it is not the truth that the signing system was changed to match that of MATPOWER regards to the OPF.

The bus table in MATPOWER is also in the consumer system. The generator matrix however is in the generator system, whereas pandapowre 1.6 used consumer system also for the gen/sgen tables. In pandapower 2.0, the signing system of the gen/sgen table is now also generator signing system.

> Second, the total cost, i.e, net.res_cost is negtive in pandapower 2.0.

Looks like the cost are negative in this data set. I suspect it was missed to change the signing of the cost values in the transation from 1.6 to 2.0. If you change the sign of the costs:

> net.poly_cost.cp1_eur_per_mw

The costs are then positive. @SteffenMeinecke  can you please check why the costs are negative in this data set?

TianmengWhu, 2019-05-07:
> I find it is not the truth that the signing system was changed to match that of MATPOWER regards to the OPF.  First, the it is said that "Bus power values are given in the consumer system.
> 
> The bus matrix in MATPOWER is also in the consumer system. The gen matrix however is in the generator system, whereas pandapower 1.6 used consumer system also for the gen/sgen tables. In pandapower 2.0, the signing system of the gen/sgen table is now also generator signing system.
> 
> Second, the total cost, i.e, net.res_cost is negtive in pandapower 2.0.
> 
> Looks like the cost are negative in this data set, as values in net.poly_cost.cp1_eur_per_mw are negative. I suspect it was missed to change the signing of the cost values in the transation from 1.6 to 2.0. If you change the sign of the costs:
> net.poly_cost.cp1_eur_per_mw *= -1
> pp.runopp(net)
> print(net.res_cost)
> 
> The costs are then positive. @SteffenMeinecke  can you please check why the costs are negative in this data set?

I also find this. When I run the code:
`import pandapower.networks 
import pandapower as pp 
net = pandapower.networks.case30()
net.poly_cost.cp1_eur_per_mw *= -1
net.poly_cost.cp2_eur_per_mw2 *= -1
pp.runopp(net)
print(net.res_cost)`

The cost is 191.99, different from the result 192.06 caculated by MATPOWER 6.0 and PYPOWER 5.1.4. Especially, the generation output are seriously different.

friederikemeier, 2019-05-07:
That looks like the boundaries and tolerances maybe differ as well.

SteffenMeinecke, 2019-11-06:
Since this issue was still not closed, I compared the powerflow and optimal powerflow of matpower 6.0 case30 and pandapower 2.1.0 case30.
Here is the pandapower code I used to visually compare the pandapower OPF results with matpower:

`import pandapower as pp`
`import pandapower.networks as pn`
`net = nw.case30()`
`for elm in pp.pp_elements():`
`    if isinstance(net[elm], pd.DataFrame):`
`        net[elm] = net[elm].sort_index()`
`pp.runopp(net)`
`net.res_ext_grid.index = net.ext_grid.bus`
`print(net.res_ext_grid.sort_index())`
`net.res_gen.index = net.gen.bus`
`print(net.res_gen.sort_index())`
`print(net.res_line.loc[[9, 34]])`
`s_from_mva = (net.res_line.p_from_mw**2 + net.res_line.q_from_mvar**2)**0.5`
`s_to_mva = (net.res_line.p_to_mw**2 + net.res_line.q_to_mvar**2)**0.5`
`s_mva = pd.DataFrame([s_from_mva, s_to_mva], index=["s_from_mva", "s_to_mva"]).T`
`print(s_mva.loc[[9, 34]])`

I also run `runpm_ac_opf` instead of `runopp`.

As stated before, the found costs slidely differs: 576.89 (Matpower) 578.49 (runopp), 577.14 (runpm_ac_opf).
The generator active power setpoints differs slidely, too: maximal difference is < 2 MW
The generator reactive power setpoints differs significantly - between all three versions -> it is a matter of the solver how to set q setpoints which have no costs.

Having a look to the maximal loaded lines (indices 9 and 34 in pandapower), using runopp the current is used to limit the power flow of the lines. Since at line 9 the voltage is < 1.0 pu, the maximum apperant power is only 31.845 MVA, not 32 MVA. At line 34, the voltage is > 1.0 pu, so the maximum apperant power is 16.638 MVA, not 16 MVA. Using matpower runopf or runpm_ac_opf, not the current but the power flow is restricted.

From my point of view, these both reasons [1) different solvers act differently, 2) different line loading limit models] for slidely differrent results should be enough to close this issue.

jhmenke, 2020-04-09:
I will close it for now due to Steffens comment, but it can be reopened if the issue is still relevant

closed 2020-04-09 by jhmenke