e2nIEE/pandapower #381: Sensitivity Matrix Calculation in Pandapower

rahulkhatri121, 2019-05-13:
Hello All (Specially PandaPower Developers),

I am a Master Thesis Student , and currently involved in a project/research of Simulation of Distributed Inverter Control Volt/Var Compensation Techniques for a real grid of Distribution System Operator.  We have modeled the network as Pandapower Network and running the power flows in advent of High Penetration of Renewable Energy Sources into Distribution Network. To investigate how effective Reactive Power Compensation (Absroption or Injection of Q using sgen) for our network is, I came across a method that require **Reactive Power Sensitivity (dV/dQ)** of Bus where I am connected my PV Inverter (RES).

After running the power flow from ,I get   Jacobian Matrix  from which in Sensitivity Matrix can be found by computing inverse of it. But my network has more than 400 Buses, the number of elements in matrix is large. I want to find this dV/dQ for only some particular buses, how can i extract this value using Bus Index ?

Your help will be highly regarded.

Thanks :)

jhmenke, 2019-05-14:
look at the pd2ppc["bus"] dict. it can be used to convert pandapower bus indices to internal ppc indices (which are used in the Jacobian)

rahulkhatri121, 2019-05-14:
Hi Jan ,

Thank you very much for the reply, 

We are running PandaPower version 1,6,0 and to understand this problem I have loaded example network 'example_multivoltage()' which have 57 Buses. After running the Power flow , i can see **_pd2ppc_lookups** dictionary not pd2ppc or its the same you are talking about. 

When I open the the dictionary , I can see the array 'bus ' which in this case has 60 values  , are these value are bus indices that can be translated to PPC , if yes , then how can i use the internal ppc indices to extract jacobian value ? as ppc indices order /size is confusing for me.

Thank you 

Thanks

lthurner, 2019-05-14:
Check out this tutorial on the internal data structure for an explanation: https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb

rahulkhatri121, 2019-05-18:
Hi @lthurner ,

Thanks , it really worked for me to extract the voltage and power angle sensitivity values using Jacobian matrix after running power flow.

However , I have a slight issue regarding the result ,

I created a small network containing of MV(10Kv) and LV Buses(0.4kV) buses with lines and transformer. To be more specific , I found of P/V sensitivity value for a LV Bus to be : -0.044709272120892075 , which if I understand right, tells by adding 1 MW of Load , voltage will reduce by this bus by this number or voltage will increase by this number when power is negative (i-e generation).

So at this bus my voltage before adding the Load Was : 0.9878901488812388

Voltage after 1MW , using sensitivity value : 0.9878901488812388--0.044709272120892075 =0.9431808767603468

Now to verify, I add the load of 1 MW , and run the power flow and res_bus returns the voltage : 0.9303379965502067 which is different then expected ? 

I have been trying this on various networks and getting the same issue where there is always difference of some decimal places between expected voltage value and value after power flow.

Any clue or reason for this little change of Decimal Points ?

Thanks

wangzhenassd, 2019-05-18:
The reason is probably the sensitivity is at nature not linear. The result is only accurate within a small range of working point deviation. @rahulkhatri121

rahulkhatri121, 2019-05-18:
Hi @wangzhenassd 

Thanks for the clarification. Do you have any clue , how to account these sensitivities non-linearity for finding out the  accurate voltages by (dV/dQ and dV/dP) ?? , as accurate voltages will result in robust PQ Control. 

Thanks

wangzhenassd, 2019-05-18:
Hi @rahulkhatri121,
I think, it's hard to quantify the non-linearity, it should be accurate enough to use jacobian matrix as a first guess, which should then be updated repetitively if you want to use it in a controller.

rahulkhatri121, 2019-05-21:
Yes, may be it can be considered as limitation, however thank you very much :)

friederikemeier, 2019-05-21:
I agree with @wangzhenassd . From my experience controllers iteratively updating the sensitivity converge nicely. One of my colleagues even once made a convergence proof for the P-control and I am sure, that this applies to Q-control as well. From the jacobian you can only get a difference quotient ΔV/ΔP , but not the true non-linear derivation ∂V/∂P.

Pampeman, 2019-05-26:
Hi rahulkhatri121 or someone else,
the extraction of the jacobian Matrix worked also for me but how can you identify the sensitivity factors for certain buses in a Network ?
I know there is a Tutorial for the extraction of the admittance and jacobian matrix under:
https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb
I tried the indications on the tutorial for getting an element of the matrix. For the me is the idenficataion of Elements on the admittance really clear but not for the jacobian.
How can you identify the voltage and power angle sensitivity values on the jacobian by indicating the index of the bus of your interest ?
Because, if you specify the index of the bus by giving ( for example) :
Ix=net._pd2ppc_lookups["bus"][ 11]
(net._ppc["internal"]["J"].todense()[Ix])
you will get a vector with mutiple values like this:
matrix([[ 0. , 0. , 0. , 0. , 0. ,
0. , 0. , 0. , 0. , 0. ,
-0.19020609, 0.19020609, 0. , 0. , 0. ,
0. , 0. , 0. , 0. , 0. ,
0. , 0. , -0.76097525, 0.80374049]])
which of those values in this vector are the voltage and power angle sensitivity values?
The jacobian Matrix in my case has the dimensions:
'J': <24x24 sparse matrix of type '<type 'numpy.float64'>'
with 136 stored elements in Compressed Sparse Row format>,
I appreciate your answer and your help
Pampeman @rahulkhatri121 @friederikemeier @lthurner

friederikemeier, 2019-05-27:
Hi @Pampeman! 

For better understanding of the dimensions of the Jacobian have a look at this paper by @FlorianShepherd : https://arxiv.org/abs/1804.06742
Especially eq. 22-26.

Hope this helps!

Pampeman, 2019-05-27:
Hallo Frau Maier,

vielen Dank für Ihre Antwort.

Ich schaue mir das Dokument..

Gruß
AA

rahulkhatri121, 2019-05-29:
Hi @Pampeman ,

I am sorry for answering you late. 

Here is how I calculated/extracted the sensitivities values from the Jacobian Matrix.

- I found Inverse of Jacobian Matrix using **(np.linalg.pinv(J))** , this is your sensitivity matrix which have order of n by n. 
- This order n= 2 time number of PQ Buses + Number of PV Buses in the network.
- In my network, I only had PQ Buses , so my order was 2 time number of Buses .
 But remember that number of Buses here means , Internal PPC buses (hence PPC index is of importance here ) , not the Pandapower Index.
- I made my network in a way that, my feeder bus is always the last bus as per pandapower index and also in Internal PPC index.
- Once you know PPC index for the bus whose sensitivity you are interested in, you can use this index , to extract value from your sensitivity. 
- Sensitivity matrix has this structure .  

     Sens =      d(Angle)/dP          d(Voltage)/dP
                      d(Angle)/dQ          d(Voltage)/dP

Here is the piece of code, that might help you to understand the implementation . 

# Mapping the Buses
panda_index=7    #Let suppose I want to find the sensitivities values at Bus 7 (Pandapower power index)
ppc_index = net._pd2ppc_lookups["bus"][panda_index]

#Select Bus whose Sensitivity With its Own Parameters
bus=ppc_index 

print(net.res_bus["vm_pu"][panda_index])

# Find the Diagonal Sensitivities
sens_p_delta=sens.item(bus,bus)
sens_p_v=sens.item(bus,(n_buses)+bus)
sens_q_delta=sens.item((n_buses)+bus,bus)
sens_q_v=sens.item((n_buses)+bus,(n_buses)+bus)

#Saving the four values in Matrix 
sens_matrix_diag=np.array([[sens_p_delta,sens_p_v],[sens_q_delta,sens_q_v]])

Note this code extract each bus sensitivities with its own change in power and reactive power , however you can definitely find one bus sensitivity value with other bus's change in power or reactive power.

I hope this helps ;)

Cheers,

Rahul

lthurner, 2019-05-29:
Thanks Rahul. Would you like to extend the jupyter notebook tutorial about the internal datastructure to include this useful information?

rahulkhatri121, 2019-05-30:
Hi @lthurner 

Yes, I woudl like to do that. But so far I had network containing PQ Buses only , as order was easy to find, but when PV buses are present, then order of Sensitivity Matrix is different, when I come across a general algorithm to extract the matrix for any network, I will extend the tutorial. 

For this do you have any idea , is there any way we could identify PV buses easily in Internal PPC in Pandapower, if there is then it will be much easier task.

Thanks

Pampeman, 2019-05-30:
Thank you Rahul for the great help.

Do you know this Document ?
https://arxiv.org/abs/1804.06742

This document was suggested from friederikemeier. I tried to read it but it´s to dense for me.

Maybe on this document you can find some Hints ( Pseudo Code) to identify the PV and PQ Buses. Because your are more involved on this Issue.
Please let us now if you have found something new About the Sensitivity matrix.

We will appreciate it !!!!!!!

lthurner, 2019-05-30:
> For this do you have any idea , is there any way we could identify PV buses easily in Internal PPC in Pandapower, if there is then it will be much easier task.

Yes there is a function that gives you the slack, pv and pq buses: https://github.com/e2nIEE/pandapower/blob/develop/pandapower/pypower/bustypes.py#L20

Pampeman, 2019-05-31:
Hi Rahul,

im still having some Questions regarding to your piece of Code as example for the extraction of sensitivity Matrix.

you wrote:

#Select Bus whose Sensitivity With its Own Parameters
bus=ppc_index

print(net.res_bus["vm_pu"][panda_index])

1) what should be the result or how should I understand the output of:
print(net.res_bus["vm_pu"] )

You wrote:
2) sens_p_delta=sens.item(bus,bus)
sens_p_v=sens.item(bus,(n_buses)+bus)
sens_q_delta=sens.item((n_buses)+bus,bus)
sens_q_v=sens.item((n_buses)+bus,(n_buses)+bus)

where does "sens" come from( see above)?
Where did you define "sens" (see above)?

It would be very helpful if you could extend the tutorial for the extraction of sensitivity Matrix.

Thxs
Pampeman

rahulkhatri121, 2019-05-31:
Hi Pampemam ,

1) Please ignore : print(net.res_bus["vm_pu"] )
as this was just for my own understanding. I was printing the values of voltages to see some results. 

2) Here is how 'sens' comes, which is actually a matrix.

J=net._ppc["internal"]["J"].todense() 
sens=(np.linalg.pinv(J))
n_buses=int((net._pd2ppc_lookups['bus'].max()))

Sorry the code I provided was very raw , which i was using for testing purpose. 

I would definitely extent the Internal Datastructure Tutorial and add the sensitivity matrix calculation , one I am able to make a general code for extraction of matrix even for networks having PV buses as well.  So  I need some time, as I am also busy with my master thesis as well. 

Thanks

Regards,

Rahul

Pampeman, 2019-06-01:
Hello Rahul,

thank you for your last answer…

I have a very last Question for you and then I will leave you alone 😊 with your master thesis !!

Suppose I have this toy example:

3 Buses

bus1 = pp.create_bus(net, vn_kv=20., name="Bus 1")
bus2 = pp.create_bus(net, vn_kv=0.4, name="Bus 2")
bus3 = pp.create_bus(net, vn_kv=0.4, name="Bus 3")

net._pd2ppc_lookups['bus']
Out[168]: array([0, 1, 2])

#Jacobian
net._ppc["internal"]["J"].todense()

matrix([[ 6.91351621, -0.27185022,  4.07060175, -2.47751891],
        [-0.34475667,  0.34475667, -2.35913046,  2.26041041],
        [-4.10659819,  2.38942136,  6.85289513, -0.28187329],
        [ 2.37999576, -2.37999576, -0.34173421,  0.25380372]])

I just want to know which  elements of the above shown jacobian corresponds to:

∂P/ ∂Vm
∂Q /∂Vm

for the BUS 2?

If I have 3 Buses  why do get a Jacobian (4x4) ?

Thank you very much!

Pampeman

P.S.

I will appreciate if one of the developers would help!!!!!

Pampeman, 2019-06-02:
Hi Mr. Thurner,

regarding to the function you mentioned to identify the PV and PQ Buses:

def bustypes(bus, gen)

In which form shlould I Input the bus/gen data?
what do you mean whit bus/gen data ?

do you mean "net.bus"  ?

or do you mean

net._pd2ppc_lookups["bus"] ?

Thank you

where does the Input come from ?

lthurner, 2019-06-02:
> In which form shlould I Input the bus/gen data? what do you mean whit bus/gen data ? 

This is a pypower function, so it referrs to the ppc, not the net. You call it with the bus and gen matrices of the ppc or ppci:

```
bustypes(ppc["bus"], ppc["gen"])
```

Pampeman, 2019-06-02:
I understand …

great! Thank you!

I hope youll have a nice sunday

lthurner, 2019-06-02:
> If I have 3 Buses why do get a Jacobian (4x4) ?

You only get the sensitivites for PQ-buses, because the voltage at PV/slack buses is fixed, so there can be no change and therefore no sensitivity.

For 2 PQ-buses, the dimension is 4x4 because you get sensitivities for P and Q respectively. Check the code here to see what quadrant referrs to which sensitivity: https://github.com/e2nIEE/pandapower/blob/6bc43ba5647dfddf3a0d9d141fa97e166ce54f47/pandapower/pf/create_jacobian.py#L41

rahulkhatri121, 2019-06-02:
Hi @Pampeman 

Answer to your questions

1.  **If I have 3 Buses  why do get a Jacobian (4x4) ?** 

Generally the order of Jacobian Matrix/Sensitivity Matrix in a Power system is given by : 
                          Order = 2*(Number of PQ Buses)+ (Number of PV Buses) .

As @lthurner, you get voltage sensitivities only at PQ Buses  , thats why you don't see a multiplication of 2 with number of PV buses, as in this case you have only voltage angle as unknown.

Coming back to your example :

Since you have defined three buses in you toy example , in which one of the bus is Feeder Bus i-e Slack Bus (at which external Grid is connected ). Now you have 2 more buses which are PQ Buses (as long as you didn't connect a generator and Voltage is unknown , then it would be a PV buses as both power and voltage will be known. ) . 

So order is simply 2 time (number of PQ Buses i-e ) is 4.

2 . **I just want to know which  elements of the above shown jacobian corresponds to:**
∂P/ ∂Vm
∂Q /∂Vm

for the BUS 2?

To know this , first find the internal index (PPC Index), which in your case might be 0,1,2. Now if you have made Bus 1 as your slack bus , then PPC index might be 0. And remaining two buses with ppc indexes 1 and 2 are your PQ buses , from which you jacobian is made.

So For bus 2 in your Jacobian matrix, the values are :

∂P2/ ∂Vm1(0,3) = 4.07060175 , 
 ∂P2/ ∂Vm2(0,4) = -2.47751891
∂Q2/ ∂Vm1 (2,3)= 6.85289513, 
 ∂P2/ ∂Vm2 (2,4)= -0.28187329

Attached in the picture you can see how jacobian matrix is ordered
[image]

Hope it helps.

Cheers, 

Rahul

Pampeman, 2019-06-02:
Hi Rahul,

Thanks….. I understood and it helped me a lot.

On my Network I have also PQ buses. I think

∂P2/ ∂Vm2 (2,4)= -0.28187 ( you meant ∂Q2/ ∂Vm2 (2,4) instead ∂P2/ ∂Vm2 (2,4) ?)

Thank you again

[Jacobian]<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F45812787%2F58759662-e431c400-852d-11e9-8950-82f63e45da24.jpg&data=02%7C01%7C%7C7e0fa8a4d7334fe1799f08d6e74161d4%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636950665326889402&sdata=Vi1%2F3tlAzY3uNDH9FzhSr8G2eGZw1Ef%2BYGghxkh4dyQ%3D&reserved=0>

closed 2019-06-03 by lthurner

AnkurArohi, 2022-11-04:
@rahulkhatri121 @lthurner @friederikemeier 

The next part of this discussion. Suppose I want to find the change in reactive power at a fixed bus due to all reactive power changes in the controllable reactive power equipments in the network.

I was anaysing the Jacobian Matrix, lets say at node "n" I would like to find the reactive power potential so to speak the maximum and minimum reactive powers possible at this point  at different network operational Reactive power values at different buses.

Looking at the matrix I am thinking of doing following pseudo code operations

1. Extract the Jacobian Matrix for a NR converged network.
2. Extract the bus indices where there are Reactive power controllable equioments
3. For the fixed active power determine the maximum stable voltage difference possible from the current value, considering the reactive power control (using Q(V) equation for a fixed P).
4. For previously defined del(V) both positive and negative use the jacobian ratios to find the del(Q) at the nth node 
`del(Qn) pos= Jacobianfactors{all controllable nodes} node*del(V)pos`
`del(Qn) neg= Jacobianfactors{all controllable nodes} node*del(V)neg`

Do you have other better suggestions, do you think this is a doable approach, any limmitations?

Highly appreciate your reply.