e2nIEE/pandapower #383: Finding the Nearest Transformer

rahulkhatri121, 2019-05-21:
Hi All ,

I have a network , in which I want to find the effect of adding the sgen element to nearest transformer connected. For that could any one of you have an idea that how I find the nearest transformer(index) which is connected to sgen bus  ?  

Thanks in advance :)

ascheidl, 2019-05-21:
Hi,

you can use the topology package. See https://pandapower.readthedocs.io/en/v2.0.1/topology.html
or
https://www.youtube.com/watch?v=ubpHrrMA63A&list=PLWIXcESdYmbxuDXSrARFZaqU3qR3JbGR5&index=9

In your case, with
`d = pd.DataFrame([top.calc_distance_to_bus(net, b) for b in net.trafo.lv_bus.values])`

you can find the nearest transformer of bus 123 with `d[123].idxmin()`

cheers
alex

rahulkhatri121, 2019-05-21:
Hi Alex,

Thanks alot ;)

ascheidl, 2019-05-21:
had to edit the code ...  first version gave the lv_bus of the transformer ... now it should give the index

closed 2019-05-24 by rahulkhatri121