e2nIEE/pandapower #386: Height and Width Adjustment of Plotly Dashboard Plot

rahulkhatri121, 2019-05-24:
Hi All ,

I am plotting my network in plotly plots and also adding the bus & line traces. As expected I am receiving the output in a HTML file that is opening directly in my browser, but width of the plots(plotly dash plot) is small .I can successfully change the width by manually edititng the HTML page source code, but obviously this is not required. 

So does anyone of you have an idea that how can we hardcore width of set size, so that every time it generates plots, it should be set width and height.

Your response and help would be really appreciated. 

Thanks & Cheers,

Rahul

Pampeman, 2019-05-27:
Hi rahulkhatri121 or someone else,
the extraction of the jacobian Matrix worked also for me but how can you identify the sensitivity factors for certain buses in a Network ?
I know there is a Tutorial for the extraction of the admittance and jacobian matrix under:
https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb
I tried the indications on the tutorial for getting an element of the matrix. For the me is the idenficataion of Elements on the admittance really clear but not for the jacobian.
How can you identify the voltage and power angle sensitivity values on the jacobian by indicating the index of the bus of your interest ?
Because, if you specify the index of the bus by giving ( for example) :
Ix=net._pd2ppc_lookups["bus"][ 11]
(net._ppc["internal"]["J"].todense()[Ix])
you will get a vector with mutiple values like this:
matrix([[ 0. , 0. , 0. , 0. , 0. ,
0. , 0. , 0. , 0. , 0. ,
-0.19020609, 0.19020609, 0. , 0. , 0. ,
0. , 0. , 0. , 0. , 0. ,
0. , 0. , -0.76097525, 0.80374049]])
which of those values in this vector are the voltage and power angle sensitivity values?
The jacobian Matrix in my case has the dimensions:
'J': <24x24 sparse matrix of type '<type 'numpy.float64'>'
with 136 stored elements in Compressed Sparse Row format>,
I appreciate your answer and your help

Pampeman

Hi @lthurner ,
Thanks , it really worked for me to extract the voltage and power angle sensitivity values using Jacobian matrix after running power flow.
However , I have a slight issue regarding the result ,
I created a small network containing of MV(10Kv) and LV Buses(0.4kV) buses with lines and transformer. To be more specific , I found of P/V sensitivity value for a LV Bus to be : -0.044709272120892075 , which if I understand right, tells by adding 1 MW of Load , voltage will reduce by this bus by this number or voltage will increase by this number when power is negative (i-e generation).
So at this bus my voltage before adding the Load Was : 0.9878901488812388
Voltage after 1MW , using sensitivity value : 0.9878901488812388--0.044709272120892075 =0.9431808767603468
Now to verify, I add the load of 1 MW , and run the power flow and res_bus returns the voltage : 0.9303379965502067 which is different then expected ?
I have been trying this on various networks and getting the same issue where there is always difference of some decimal places between expected voltage value and value after power flow.
Any clue or reason for this little change of Decimal Points ?
Thanks
Originally posted by @rahulkhatri121 in #381 (comment)

rahulkhatri121, 2019-07-07:
Issue solved by doing some changes in the internal plotly library.

closed 2019-07-07 by rahulkhatri121