e2nIEE/pandapower #388: R and X Calculation from the Ybus Matrix -

rahulkhatri121, 2019-06-03:
Hi All and Pandapower Developers,

I want to find the R and X values at one of my Bus where I am connecting the DG so that I can use this equation basically to calculate the change in voltage that occurs because of reverse power flow due to DG.

My Simple test network looks like this : 

[image]

Here for some reasons I want to connect my DG on Bus 2, then my objective is to find the Change is voltage dV from b0 to b2 and then to calculate I have to use this equation : 

[image]

Now a question ,how to find the R and X values between bus 2 and bus 0 to compute this change in voltage. 

 I tried inversing the admittance matrix after power  to find the Zbus matrix and then 

a. Used  **element (0,2)**, is this the impedance matrix which if I am right is impedance between bus 0 and bus 2

b.  I also used self impedance of Bus 2 from Zbus=Inverse(Ybus) t, which is impedance of Bus to with whole network

I kind of tried using both impedance values  to compute the change in voltage occuring from bus0 to bus2 at certain change in Power(KW), and to verify I also run the pandapower power flow, by changing same amount of power on the bus, unfortunately change in voltage is different from computed. 

May be I am doing a mistake somewhere.

Your help or insight on this would be really appreciated.

Thanks

Cheers,

Rahul

p.s why I am doing this ?
 I want to design a Reactive Power based on Voltage Q(U) control strategy for my DG Inverter unit. I want to set a dynamic droop control which in which voltage thresholds are set based on the measured Vpcc voltage and R and X values at the bus where DG is being connected.  Also your response whether it a suitable method or not . ?

@jhmenke @lthurner @friederikemeier

jhmenke, 2019-06-03:
>a. Used element (0,2), is this the impedance matrix which if I am right is impedance between bus 0 and bus 2

since buses 0 and 2 are not connected (assuming pandapower bus index equals ppc bus index in this case), the admittance of element (0, 2) will probably be 0.

i don't know about the other points, but using an equation which suggests a linear relationship between power, impedance and voltage drop seems inherently imprecise to me. where is the equation from, by the way?

best of luck

rahulkhatri121, 2019-06-03:
Hi @jhmenke 

Thanks for the reply.

Bus 0 and 2 are electrically connected by lines in between , and I can se that element (0,2) is not zero is admittance matrix, rather there is some value of R and X . Actually all the values of R and X are very close to each other , as I have the same length of lines.

This is how my admittance matrix looks like , from which I can see no element is 0 , as we have all electrically connected somehow.

[image]

About the Equation :

It come from here, 
[image]

This is the general relation of voltage change in a distribution network with change in power and reactive power. 

Thanks

closed 2019-06-04 by rahulkhatri121