e2nIEE/pandapower #393: No module

heinlmic, 2019-06-17:
Hello,
I have installed latest version of PP and I have tried import module:
`import pandapower.networks as nw`

but I get error
```
import pandapower.networks as nw
ModuleNotFoundError: No module named 'pandapower.networks'; 'pandapower' is not a package
```

What am I doing wrong?

lthurner, 2019-06-17:
Do you have multiple Python versions installed and maybe installed it in a different python installation / environment?

heinlmic, 2019-06-18:
Thanks for advice.
Solution was very easy and I am stupid. I named my project pandapower.py and that was the biggest problem.

closed 2019-06-18 by heinlmic

agaa2, 2020-12-05:
Can someone help me? I have the same problem ModuleNotFoundError: No module named 'pandapower', In which environment should I install pandapower? I get only python 3, and I installed pandapower in pypsa environment