e2nIEE/pandapower #394: line impedence

Prashantgrid, 2019-06-17:


lthurner, 2019-06-17:
There is no function that does this, but you can use the topology searches for that. Networkx can find the paths and then you can calculate the impedance from that. This should point you in the right direction:

```
import pandapower.networks as nw
import pandapower.topology as top
import networkx as nx
net = nw.create_cigre_network_hv()
mg = top.create_nxgraph(net)
for path in nx.all_simple_paths(mg, 3, 11):
    lines = top.lines_on_path(mg, path)
    print("path with resistance %.2f"%net.line.r_ohm_per_km.loc[lines].sum())
```

lthurner, 2019-06-18:
If you do implement a function that calculates the impedance between two buses that would be a very good function to add to the pandapower toolbox. Please feel free to open a pull request if you have implemented that.

FlorianShepherd, 2019-08-05:
Please feel free to reopen the issue if you have any further questions or open a pull request.

closed 2019-08-05 by FlorianShepherd