e2nIEE/pandapower #396: NaN and None in excel converter

ascheidl, 2019-06-18:
Excel has no representation of values NaN and None. So the question is how to encode these values properly.

Currently these values are saved as empty cells. Consequently, since pd.read_excel interprets empty cells as NaN, also None and empty strings end up as NaN if a networks is saved and restored from excel.

Some possible solutions:

1. save as "None" and "NaN" in excel
pro: easy to impl.
con: diverges from current behavior, names can not be the string "None"

2. save as special formulas in excel (from XlsxWriter docu: nan_inf_to_errors: Enable the worksheet.write() and write_number() methods to convert nan, inf and -inf to Excel errors. Excel doesn’t handle NAN/INF as numbers so as a workaround they are mapped to formulas that yield the error codes #NUM! and #DIV/0!)
con: diverges from current behavior, formulas might confuse users

3. keep representing as empty cell, when loading, transform empty cells of columns with dtype float to np.nan, transform empty cells of columns with dtype object to None
pro: keeps current behavior
con: empty strings and None cannot be distinguished, consequently as loaded network can differ from the saved one

vogt31337, 2025-07-23:
@ascheidl, still relevant?

KS-HTK, 2025-08-13:
Seems to be still relevant.
```python
from pandas import DataFrame
from pandapower.create import create_empty_network
from pandapower.file_io import to_excel, from_excel

net = create_empty_network(name='test', f_hz=55., sn_mva=1.2, add_stdtypes=False)

net['test'] = DataFrame(data={'string': ['None', ""], 'float': [float('nan'), float('inf')], 'None': [None, None]})

to_excel(net, 'test.xlsx')

net_import = from_excel('test.xlsx')
```

net.test:
 &nbsp; | string | float | None
---|---|---|---
0 | None | NaN | None
1 |  | inf | None

net_import.test:
 &nbsp; | string | float | None
---|---|---|---
0 | None | NaN | NaN
1 | None | inf | NaN

closed 2025-11-26 by KS-HTK