e2nIEE/pandapower #398: Improvement for linear powerflow internals

Gseguelg, 2019-06-18:
I've seen that the _ppc['internal'] dict for (at least) incorporated 39 bus case system after running a linear/dc power flow, also comes with keys: 'Ybus', 'Yf', 'Yt'. But it does not has susceptance matrix directly available. Not even the complete Ybus matrix, infact this last one is empty.

```
>>> import pandapower as pp
>>> from pandapower.networks import case39
>>> net=case39()
>>> net._ppc
>>> pp.rundcpp(net)
>>> net._ppc['internal'].keys()
dict_keys(['Ybus', 'Yf', 'Yt', 'branch_is', 'gen_is', 'DLF', 'buses_ord_bfs_nets', 'ref_gens'])
>>> net._ppc['internal']['Ybus']
array([], dtype=complex128)
```

I've tested it with versions 1.6.1 and 2.0.1. Both yield to same results.
I my opinion, this B matrix should be easly available, specially when it's called 'internal'.

Note: This matrix can be found by excecuting a normal powerflow and get the negative imaginary part of Ybus, but running two powerflows to get it same results doesn't look right.

lthurner, 2019-06-21:
Yes I agree that the B matrix used in the linear power flow should be saved in the "internal" dict just as the Ybus matrix is saved for non-linear power flow.

However what you are describing here:

> Note: This matrix can be found by excecuting a normal powerflow and get the negative imaginary part of Ybus, but running two powerflows to get it same results doesn't look right.

is the [B matrix](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/pypower/makeB.py#L15) which is used for the decoupled power flow. The linear power flow uses the linearized version of that, which is called the [Bdc matrix:](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/pypower/makeBdc.py#L25).

I would propose to save the B matrix into the "internal" dict inside this function: https://github.com/e2nIEE/pandapower/blob/2d437e56dd3684de2d473b18cda64e7c9f91b02e/pandapower/pf/run_dc_pf.py#L20. Do you want to put in a PR with this change?

Gseguelg, 2019-06-22:
As far as I remember **Bp** (from DLF) should return the same as **Bdc**. But sure, I will put a PR for that using **Bbus** output from [Bdc matrix](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/pypower/makeBdc.py#L25) function, as it's the official for the method.

Now the question will be: include Bbus into the existing key `net._ppc['internal']['Ybus']`? or a new one?

lthurner, 2019-06-24:
> Now the question will be: Will it be enough with including a new key Bbus only for linear power flow? or it's necessary to update all the structure of internal keys?

Adding a new key Bbus is fine

closed 2019-06-25 by lthurner