e2nIEE/pandapower #399: Short circuit current calculation

heinlmic, 2019-06-19:
Hello,
It is possible to get short circuit current in branch just for fault in one bus?
If I understand well, PP compute fault in all busses and MIN or MAX current in branches, but can I get fault current in branches for fault in one Bus?

Thank you.

Michal Heinl

lthurner, 2019-06-21:
That is currently not possible, but it would be possible to implement this. pandapower currently takes the maximum/minimum of the branch currents for all faults, so at some point in the code you have all currents calculated. Should be possible to at this point output the current for one fault instead of max/min of all faults. I would have to look into the code more exactly to figure at where and how this would be possible

heinlmic, 2019-06-23:
If will be possible to get all currents (not just min max), as you mentioned, it will be the best what I cen get.
Thank you.

lthurner, 2019-06-24:
I have added the function sc.calc_sc that takes a bus as an input and calculates the short circuit for this specific bus. 

I have written some [tests](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/test/shortcircuit/test_sc_single_bus.py#L17) that also serve as examples/documentation for now.

The results are saved in the power flow style but with the _sc suffix. so the res_bus_sc table shows the voltage profile in the short-circuit case (net.res_bus_sc.vm_pu, net.res_bus_sc.va_degree). I haven't found any external reference to validate this against, but it seems plausible on the first glance (voltage is zero at the fault bus and grows with distance from the short circuit).

The line table shows the currents through the branches (net.res_line_sc.i_ka,  net.res_line_sc.loading_percent etc.), transformers work accordingly. I have validated the branch currents against the vectorized SC current calculation (the sum of alle currents flowing into the fault bus is the same current that is given as a fault current for this bus in the vectorized version). Validation against a different tool would still be a todo as well.

@heinlmic pleases check out if this works for you and covers your use case

heinlmic, 2019-06-27:
Thank you, I compared it with other software and so far are rusults correct.

closed 2019-06-27 by heinlmic