e2nIEE/pandapower #401: 'dict' object has no attribute 'version'` loading the network 1.6.1 from json

Xevib, 2019-06-20:
If I load a network from the version 1.6.1 stored on a JSON fails on this 

``` python
pandapower.from_json(s)
```

At this point the net variable is a dict and crashes  with the error `AttributeError: 'dict' object has no attribute 'version'`

https://github.com/e2nIEE/pandapower/blob/develop/pandapower/convert_format.py#L18

lthurner, 2019-06-21:
That is strange, it shouldn't come out as a dictionary. There are tests that load networks saved in older versions to check backwards compatibility: https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test/test_files/old_versions

Can you check if you can load the file that is saved there with 1.6.1?

lthurner, 2019-09-02:
Closing for inactivity, please provide a reproducable example if this issue is still not resolved.

closed 2019-09-02 by lthurner