e2nIEE/pandapower #408: Switch with short circuit cal.

heinlmic, 2019-06-27:
Hello,
I have done some calculation and it seems that switches are not involved with short circuit calculation.

```
print(net.switch)
    bus  element   et type  closed name  z_ohm
Branch results are in beta mode and might not always be reliable, especially for transformers
0     5        0  NaN    l    True  NaN    0.0
1     5        7  NaN    l    True  NaN    0.0
2     6        2  NaN    l    True  NaN    0.0
3     6        8  NaN    l    True  NaN    0.0
**4     4        1  NaN    l   False  NaN    0.0**
5     4        6  NaN    l    True  NaN    0.0
6     7        7  NaN    l    True  NaN    0.0
7     7        8  NaN    l    True  NaN    0.0
8     7        6  NaN    l    True  NaN    0.0
9     5        5  NaN    l    True  NaN    0.0
10    5        4  NaN    l    True  NaN    0.0
11    4        4  NaN    l    True  NaN    0.0
12    4        3  NaN    l    True  NaN    0.0
13    6        3  NaN    l    True  NaN    0.0
14    6        5  NaN    l    True  NaN    0.0
**15    2        1  NaN    l   False  NaN    0.0**
```

Element 1 in rows 15,4 is Flase

```
sc.calc_sc(net,case = 'min',branch_results=True)
print(net.res_line_sc)

    ikss_ka
0  0.063903
**1  0.063903**
2  0.063903
3  0.047927
4  0.047927
5  0.047927
6  0.015976
7  0.015976
8  0.015976
```
in that case there should be zero ikss in line 1.

Am I right or not?

Best regards,
Heinl

lthurner, 2019-07-08:
I wasn't quite able to reproduce this. With this test network:

```
import pandapower.shortcircuit as sc
import pandapower as pp

net = pp.create_empty_network()
b1 = pp.create_bus(net, 20.)
b2 = pp.create_bus(net, 20.)
b3 = pp.create_bus(net, 20.)
b4 = pp.create_bus(net, 20.)

pp.create_ext_grid(net, b1, s_sc_max_mva=50, rx_max=0.1)

l1 = pp.create_line(net, b1, b2, 2., std_type="NA2XS2Y 1x95 RM/25 12/20 kV")
l2 = pp.create_line(net, b2, b3, 2., std_type="NA2XS2Y 1x95 RM/25 12/20 kV")
l3 = pp.create_line(net, b3, b4, 2., std_type="NA2XS2Y 1x95 RM/25 12/20 kV")
l3 = pp.create_line(net, b4, b1, 2., std_type="NA2XS2Y 1x95 RM/25 12/20 kV")

pp.create_switch(net, b2, l2, et="l", closed=False)
pp.create_switch(net, b3, l2, et="l", closed=False)

sc.calc_single_sc(net, b4)
print(net.res_bus_sc.vm_pu.round(4))
print(net.res_line_sc.loading_percent)
```

I ran into an error because there was still a bug in the single_sc calculation. I fixed this in #424

However the case that you showed that you see a flow through a branch that should be out of service I was not able to reproduce. Can you provide minimal example that shows this problem?

heinlmic, 2019-07-10:
I ran my test network many times, than I found out that in switch a change "et" and "type", When I fixed it type - NAN, et - "l" it is ran correct.
I am sorry to bothered you with this.

closed 2019-07-10 by heinlmic