e2nIEE/pandapower #412: Sorting the Bus Voltage as Per Electrical Distance in Radial Feeder

rahulkhatri121, 2019-07-03:
Hello All ,

I am trying to plot voltage profile of my network which has more than 400 Buses of a real distribution feeder. However in radial network branches are and in my res_bus matrix , buses are ordered the way they are entered into network. Now i want to see voltage rise/drop along the feeder from substation to last bus of the feeder (from electrical distance point of view). 

So is there any way that sort the buses as per electrical distance from the Substation(External Grid)  ?

Thanks alot for the help..

Cheeers :)

Rahul

ascheidl, 2019-07-03:
hi,

for finding electrical distances you can use our topology package, specifically the function calc_distance_to_bus. [https://pandapower.readthedocs.io/en/v2.0.1/topology.html](url)

However, for plotting the voltage profile just use our function plotting.plot_voltage_profile.

cheers,
alex

rahulkhatri121, 2019-07-07:
Thanks @ascheidl  , this feature seems helpful :)

closed 2019-07-07 by rahulkhatri121