e2nIEE/pandapower #414: Enhance pp grid datastructure with an epsg code for geodata

vogt31337, 2019-07-03:
Hi,

I just tried to convert the mv_oberrein grid (from pandapower.networks) to WGS84 (epsg=4326) coordinates. And it resulted in a grid situated in Somalia, which doesn't seem to be correct...
I think the grid is saved in another gauß-krüger format (https://spatialreference.org/ref/epsg/31467/)

```Python
from pyproj import Proj, transform

inProj = Proj(init='epsg:31467')
outProj = Proj(init='epsg:4326')

for i in grid.bus_geodata.index:
	x,y = transform(inProj,outProj,grid.bus_geodata.x[i],grid.bus_geodata.y[i])
	grid.bus_geodata.x[i] = x
	grid.bus_geodata.y[i] = y
```

So I would suggest to save in an additional field the epsg code of the coordiates, so that a change in projection system would be easily possible. Maybe enhance the api or the to_***(net, file) function, so that a user can save the grid in a different projection.

rbolgaryn, 2019-07-10:
If the net.bus_geodata and/or net.line_geodata are a GeoDataFrame and have a crs set, as they should in this case, saving and loading net to/from json keeps this information and restores it appropriately. 
After loading the grid, the type of bus_geodata is still GeoDataFrame with the crs argument properly set.

rbolgaryn, 2019-07-10:
So this is the issue with the specific net rather than the saving/loading implementation.

rbolgaryn, 2019-07-10:
A solution could be resaving all pandapower.networks grids to make sure all geodata information is implemented using GeoDataFrame with crs set.

FlorianShepherd, 2019-08-06:
@vogt31337 I think this is already fixed, right?
See: https://github.com/e2nIEE/pandapower/blob/develop/pandapower/plotting/geo.py 
`convert_epgs_bus_geodata()`

closed 2019-08-06 by FlorianShepherd