e2nIEE/pandapower #417: pp.to_json not quite json conform

vogt31337, 2019-07-04:
Hi,

saving with pp.to_json produces a json file which is not really readable with other tools. There is some extensive escaping going on.
For example mv_oberrhein saved to json:
```json
{
    "_module": "pandapower.auxiliary",
    "_class": "pandapowerNet",
    "_object": "{\"OPF_converged\":false,\"bus\":{\n    \"_module\": \"pandas.core.frame\",\n    \"_class\": \"DataFrame\",\n    \"_object\": \"{\\\"columns\\\":[\\\"name\\\",\\\"vn_kv\\\" [rest omitted]
}
```
So this is a valid json structure, but the stuff in "_object" isn't readable due to exessive escaping.
Better would be something like:
```json
{
	"_module": "pandapower.auxiliary",
	"_class": "pandapowerNet",
	"_object": {
		"OPF_converged": false,
		"bus": {
			"_module": "pandas.core.frame",
			"_class": "DataFrame",
			"_object": {
				"columns": ["name", "vn_kv", "type", "zone", "in_service"],
				"index": [0, 1, 100, 101, 102, 103, 104, 106, 107, 108, 109, 110, 111, 116, 117, 118, 119, 120, 126, 129, 131, 132, 133, 134, 136, 137, 138, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 153, 155, 157, 159, 161, 162, 167, 168, 169, 170, 171, 172, 173, 174, 176, 178, 181, 184, 186, 188, 189, 190, 192, 194, 195, 196, 197, 198, 199, 2, 200, 201, 205, 207, 210, 213, 215, 216, 219, 221, 223, 224, 227, 229, 231, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 253, 269, 271, 273, 275, 281, 285, 286, 287, 288, 289, 29, 290, 298, 3, 30, 301, 303, 304, 305, 31, 312, 313, 314, 315, 316, 317, 318, 319, 32, 33, 34, 35, 36, 37, 38, 39, 4, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 5, 50, 51, 52, 53, 54, 55, 56, 57, 58, 6, 64, 65, 7, 71, 72, 73, 74, 75, 76, 77, 78, 79, 8, 80, 81, 82, 83, 84, 85, 86, 94, 95, 98],
				"data": [
					["bus_200", 20.0, "n", null, true],
					["bus_204", 20.0, "n", null, true],
					["bus_124", 20.0, "n", null, true],
					["bus_227", 20.0, "n", null, true],
					["bus_125", 20.0, "n", null, true],
					["bus_228", 20.0, "n", null, true],
					["bus_225", 20.0, "n", null, true],
					["bus_226", 20.0, "n", null, true],
					["bus_128", 20.0, "n", null, true],
					["bus_223", 20.0, "n", null, true],
					["bus_129", 20.0, "n", null, true],
					["bus_224", 20.0, "n", null, true],
[rest omitted]
```

Maybe it's possible to correct the JSONEncoder to produce something without the escaped chars.

Thx!

lthurner, 2019-07-04:
This is already fixed, see #406

vogt31337, 2019-07-04:
Thanks!

I would vote against such a distinction, first of all, disk space isn't relevant anymore. So smaller file sizes for files being about 1 MB to 10 MB is negligible. Second, deciding what format (small / exchangable) will always lead to some frustration because some user can open a particular grid and some users can't and need to resave it.

What I've found out, the code before #406 just double jsonify's the data. So just calling json.decode on the "_object" field makes the grid readable in java (and maybe in other languages too). 

Keep in mind, that the pp decoder will be expecting this double jsonification...

lthurner, 2019-07-04:
> I would vote against such a distinction, first of all, disk space isn't relevant anymore.

File size certainly matters in some cases. For example currently the pandapower installation has 20MB, with grids saved in a different format it might grow to 100MB without any advantage for the large majority of users that don't care about the internal structure of the json file.

Also, its not just about file size but about speed. The larger the file, the longer it takes to save and load the networks.

> Second, deciding what format (small / exchangable) will always lead to some frustration because some user can open a particular grid and some users can't and need to resave it.

Obviously both versions would have to be readable with pandapower, but for external applications that is true.

> Keep in mind, that the pp decoder will be expecting this double jsonification...

Well yes, if the pp decoder wouldn't already do that you would not be able to load any data in pandapower :). The double jsonification is what I removed in #406, but you still have the same issue when going into the dataframes. Each dataframe is dropped as a json string in the _object entry. So the pandapower json format is actually a nested json File, where strings in _object might be further json-strings.

lthurner, 2019-07-08:
@rbolgaryn can you look into if there is a way to make the to_json format json conform without having significantly larger filesizes and runtime for serialization? For example by using the pd.DataFrame.to_dict function instead of .to_json in the to_serialiazable?

vogt31337, 2019-07-18:
> If that is not possible, we should at least document the json structure in the pandapower docs.

Just document it, because double deserilization is working fine.

rbolgaryn, 2019-10-25:
#528 makes it relevant again.

rbolgaryn, 2019-10-30:
Here once again a comparison of performance for json_io for using to_dict and to_json for DataFrames:

[image]

Loading is similar, but saving of large grids to json takes much longer with to_json, while the pandas implementation of DataFrame.to_json is very fast.

jhmenke, 2020-04-09:
Can this be closed?

closed 2020-05-18 by lthurner

hailiangliu89, 2023-02-19:
> Here once again a comparison of performance for json_io for using to_dict and to_json for DataFrames:
> 
> [image]
> 
> Loading is similar, but saving of large grids to json takes much longer with to_json, while the pandas implementation of DataFrame.to_json is very fast.

Hi @rbolgaryn Roman, I am experiencing the same. `s = pp.to_json(net) ` sometimes takes up to 20 minutes with large grids and long timeseries. I am wondering if there is a workaround to it. :)

rbolgaryn, 2023-02-19:
Hi @hailiangliu89 ,

I suggest dropping the ConstControl elements before saving the grid