e2nIEE/pandapower #418: create_polynomial_cost function error

Misbahrani, 2019-07-05:
Hello
I am trying to implement quadratic cost funciton, and m getting following errros:
[image]
[image]
and following:
[image]

friederikemeier, 2019-07-05:
Hi, the function was renamed in pandapower 2.0 and it is now created without the numpy array. Please check the documentation of the function!
Thats what it is now:

`create_poly_cost(net, element, et, cp1_eur_per_mw, cp0_eur=0, cq1_eur_per_mvar=0,
                           cq0_eur=0, cp2_eur_per_mw2=0, cq2_eur_per_mvar2=0, type="p", index=None)`

lthurner, 2019-07-05:
Please see also the documentation that differences between 1.6 and 2.0: https://pandapower.readthedocs.io/en/v2.0.1/about/update.html#cost-functions

Misbahrani, 2019-07-06:
Okay, thank you so much

closed 2019-07-06 by lthurner