e2nIEE/pandapower #419: Two-winding transformer: Main field equivalent circuit elements in test cases

ckittl, 2019-07-05:
Hey there,

I got a closer look at your repo and was really happy to find such sophisticated test cases for your power flow calculation - I really appreciate it!

However, while trying to understand how exactly your handling of equivalent circuit diagrams works, I stumbled upon the two winding transformer type parameters. For the no load calculation of the transformer you provide the no load current `i0_percent` = 0.4% and the iron losses with `pfe_kw` = 2.0 kW (0.5 % of `sn_mva`). Might be, that I mess up some formulae, but I think those parameters do not fit together -- there is more current flowing through `b_real` than through `b_real` || `b_img`.

I tried to validate my thoughts through the following entry point:
```python
pandapower.test.loadflow.test_runpp.test_init_results
```
and this actually leads to neglecting the imaginary part of the main field elements in
```python
pandapower.build_branch._calc_y_from_dataframe
```
in line 327.

I would be glad if you would enlighten me if I'm wrong or if that test was designed intentionally that way. The test is very valuable, because it proofs, that all tools handle it in the same way. On the other hand, I think you're testing a corner case... If you need any further, please just ask.

Best regards,
Chris

lthurner, 2019-07-05:
Huh, thats interesting, you are completely right about that. We have intentionally implemented the behaviour that when the current pfe_kw > i0_percent, the iron losses are set to zero and the whole i0_percent goes into the reactance of the transformer. That is achieved with this line, that sets the iron losses to zero in case they would become negative:
https://github.com/e2nIEE/pandapower/blob/fcd7ad1ee51aac2af318a68555c001969b86950e/pandapower/build_branch.py#L327
We have adopted that behaviour from PowerFactory.

However, you are right that this is a corner case that should be tested seperately and the main test for the transformer model should be with consistent transformer parameters.

ckittl, 2019-07-05:
Ah, sorry... I mixed up real and imaginary part. Anyway, thanks for your fast reply! :-)

lthurner, 2019-07-05:
I don't think you mixed it up, real and imaginary part are actually switched in the implementation, because the column in the MATPOWER format gives the susceptance B and doesn't provide a column for the conductance G. So internally the value becomes:

```
Y = j*B
```

To model a real component in this, we make B a complex number:

```
B = Br - j*Bi
```
which leads internally to

```
Y = j(Br - j*Bi)
   = j*Br + Bi
```

With this trick we modeled a transformer iron loss without changing the PYPOWER data format. So this is a relic from when we where much closer attached to PYPOWER. Maybe we could also let this go and just introduce a new column into the data format to make the whole implementation less confusing...