e2nIEE/pandapower #432: IndexError when creating collections

marhofmann, 2019-07-15:
A IndexError occurs when creating a collection (load, sgen, gen) if some elements have been dropped before.
```
in create_load_collection p2 = p1 + _rotate_dim2(np.array([0, size * off]), ang[i])
IndexError: list index out of range
```

With the following example it sould be possible to reproduce the error:
```
import pandapower as pp
import pandapower.networks as nw
import pandapower.plotting as plt
import pylab
import numpy as np

net = nw.simple_mv_open_ring_net()

def plot():
    bus_col = plt.create_bus_collection(net, color="k", size=0.02)
    trafo_col = plt.create_trafo_collection(net, size=0.1)
    line_col = plt.create_line_collection(net, color="k", use_bus_geodata=True)
    ext_grid_col = plt.create_ext_grid_collection(net, size=0.2)
    load_col = create_load_collection(net, size=0.1, orientation=np.pi/2)
    plt.draw_collections([bus_col, trafo_col, line_col,
                          ext_grid_col, load_col], figsize=(8, 6))
    pylab.show()

net = nw.simple_mv_open_ring_net()

plot()
print(net.bus)
print(net.line)
print(net.load)
print(net.sgen)

net.switch.closed = True
net.bus.in_service.loc[3] = False
net.line.in_service.loc[1] = False
net.line.in_service.loc[2] = False
pp.drop_inactive_elements(net)

plot()
print(net.bus)
print(net.line)
print(net.load)
print(net.sgen)
```

The reason is that ```iterrows()``` in the for loops of the ```create_xy_collection()``` runs into an empty row after deleting elements. I propose to change from ```iterrows()``` to ```range(len(table))```

FlorianShepherd, 2019-07-15:
Hello marhofmann,

thank you for pointing this out and proposing already a solution!

This line of code should be faster, since .at[] ist faster than .iloc[] I suppose. I tested and it seems to work. Maybe you can modify your solution and open a pull request. Would be great!

Best 
Florian

```
    for i, idx in enumerate(load_table.index):
        p1 = net.bus_geodata[["x", "y"]].loc[net.load.at[idx, "bus"]]
```

FlorianShepherd, 2019-07-15:
> Hello marhofmann,
>
> thank you for pointing this out and proposing already a solution!
>
> This line of code should be faster, since .at[] ist faster than .iloc[] I suppose. I tested and it seems to [...]

Ok I fixed this at multiple points -> see 37fa400

marhofmann, 2019-07-17:
Great! Looks much better. But why don't you make a PR?

FlorianShepherd, 2019-07-17:
> Great! Looks much better. But why don't you make a PR?

I did: https://github.com/e2nIEE/pandapower/pull/434
If you're ok with it, I'll decline your pull request and merge this.

marhofmann, 2019-07-17:
Of course!

closed 2019-07-18 by FlorianShepherd