e2nIEE/pandapower #436: pp.from_excel not working

vogt31337, 2019-07-18:
Hi,

Pandapower Version 2.0.1 (release) isn't able to load excel files.
```python
import pandapower as pp
import pandapower.networks

net = pandapower.networks.example_multivoltage()
pp.to_excel(net, 'congestion.xls')

net = pp.from_excel('congestion.xls')
```

Results in:
```
Traceback (most recent call last):
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\file_io.py", line 277, in from_excel
    net = from_dict_of_dfs(xls)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\io_utils.py", line 158, in from_dict_of_dfs
    net = df_to_coords(net, item, table)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\io_utils.py", line 136, in df_to_coords
    net[item].loc[:, ['x', 'y']] = table.loc[:, ['x', 'y']]
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\indexing.py", line 1472, in __getitem__
    return self._getitem_tuple(key)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\indexing.py", line 890, in _getitem_tuple
    retval = getattr(retval, self.name)._getitem_axis(key, axis=i)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\indexing.py", line 1901, in _getitem_axis
    return self._getitem_iterable(key, axis=axis)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\indexing.py", line 1143, in _getitem_iterable
    self._validate_read_indexer(key, indexer, axis)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\indexing.py", line 1206, in _validate_read_indexer
    key=key, axis=self.obj._get_axis_name(axis)))
KeyError: "None of [['x', 'y']] are in the [columns]"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\indexes\base.py", line 3078, in get_loc
    return self._engine.get_loc(key)
  File "pandas/_libs/index.pyx", line 140, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 162, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1492, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1500, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'parameters'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\test.py", line 8, in <module>
    net = pp.from_excel('congestion.xls')
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\file_io.py", line 279, in from_excel
    net = _from_excel_old(xls)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\file_io.py", line 286, in _from_excel_old
    par = xls["parameters"]["parameters"]
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\frame.py", line 2688, in __getitem__
    return self._getitem_column(key)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\frame.py", line 2695, in _getitem_column
    return self._get_item_cache(key)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\generic.py", line 2489, in _get_item_cache
    values = self._data.get(item)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\internals.py", line 4115, in get
    loc = self.items.get_loc(item)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandas\core\indexes\base.py", line 3080, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas/_libs/index.pyx", line 140, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 162, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 1492, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 1500, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'parameters'
```

teo9876, 2019-08-23:
I've a similar issue with the same release. The Error Log is not exactly the same, but similar.

Here is reported the code:

[code.txt](https://github.com/e2nIEE/pandapower/files/3534575/code.txt)

And here the log

[log.txt](https://github.com/e2nIEE/pandapower/files/3534577/log.txt)

motsi7, 2019-10-30:
Hey, I have the same issue with version `2.1.0`. Is there any update on this?

I also tried to pip install from `develop` branch but the bug persists.

@ascheidl could you help with this?

ascheidl, 2019-10-30:
I will have a look

ascheidl, 2020-01-16:
sorry for not responding for a longer time

is this still an issue ? (I can not reproduce in the current version )

ottisredds10, 2020-08-19:
I was having this same issue and realized my column name in the "parameters" sheet was named "parameter" instead of "parameters". Once I added the "s" onto the end of the column name in my Excel file the network uploaded without error.

I originally saved the network file down using pp.to_excel so it looks like the Excel "parameters" sheet is getting saved down with the column name entered wrong. During pp.to_excel the "parameters" sheet should name the column "parameters" with the "s" to avoid this issue when reloading the network file using pp.from_excel.

SteffenMeinecke, 2021-09-22:
Closed due to inactivity (@teo9876's code works with current version; @vogt31337's code works too if data type "xlsx" is used instead of "xls")

closed 2021-09-22 by SteffenMeinecke