e2nIEE/pandapower #438: create_bus_traces  type of infofunc changed?

fabbbi, 2019-07-23:
Did the required type of infofunc for plotting.plotly.create_bus_traces change?

V2.1.0:
`bus_trace['text'] = net.bus.loc[bus_plot_index, 'name'] if infofunc is None else infofunc.loc[buses]`
V2.0.1:
`bus_trace['text'] = net.bus.loc[bus_plot_index, 'name'] if infofunc is None else infofunc`

Docs and comments in code still suggest infofunc should be of type list. Thus it throws following error:
`AttributeError: 'list' object has no attribute 'loc'`

lthurner, 2019-07-24:
@jhmenke I think you updated the plotly library, can you look into this?

jhmenke, 2019-07-30:
i will fix it next week

jhmenke, 2019-08-05:
Please confirm that it works again @fabbbi

fabbbi, 2019-08-06:
It's not on pip so added it by hand. 
For me it works for that section but an AssertionError is raised as infofunc being a pd.Series is asserted later in the code:

line 305:
`    if infofunc is not None:
        if len(infofunc) != len(lines) and len(infofunc) != len(net.line):
            raise UserWarning("Different amount of hover info than lines to plot")
        assert isinstance(infofunc, pd.Series), \
            "infofunc should be a pandas series with the net.line.index to the infofunc contents"`

This happens in the create_trafo_trace function and also in the create_trafo_trace function.
Around lines: 305, 418,478

jhmenke, 2019-08-06:
Sorry, apparently i overlooked something.

In the meantime, you can comment out line 305 for your use.

jhmenke, 2019-08-06:
I am currently changing it but to be honest, allowing lists or other types of iterables other than pandas.Series makes thing vulnerable to errors.

Lists can only really be used if the infofunc is valid for all elements of the type. Selecting only a subset (e.g., excluding open lines etc.) creates issues as well as noncontinuous pandapower indices, which can happen as well.

As it stands, i want to revert the changes from the issue and just make this more clear in the documentation.

Code can be converted easily, e.g., for buses (*old_infofunc* is the list with information):
`infofunc=pd.Series(index=net.bus.index, data=old_infofunc)`

jhmenke, 2019-08-06:
Can you check PR #464 and see if that solves your problem?

commit ad56b09f (2019-08-05): Merge pull request #450 from jhmenke/develop

Fix #438 and improve estimation logger messages
closed 2019-08-12 by jhmenke