e2nIEE/pandapower #440: Calculation of R/X Ration at any Bus

rahulkhatri121, 2019-07-23:
Hello Everyone and Pandapower Awesome Developers, 

I am in quest/search of some method to calculate the R/X ratio at any point (say Bus/Node) of pandapower network, which takes care of the equivalent network impedance at that point considering all the lines and transformers. 

If you know , how we can do it , I would be really happy to know :)

Thanks

Cheers,

Rahul

lthurner, 2019-07-24:
Have you looked into the short-circuit module? It includes a calculation of short-circuit impedances at all buses: https://pandapower.readthedocs.io/en/v2.1.0/shortcircuit/ikss.html#voltage-source-contribution

FlorianShepherd, 2019-08-05:
If you have any further questions, please reopen!

closed 2019-08-05 by FlorianShepherd