e2nIEE/pandapower #441: Allow index/id to be string

bergkvist, 2019-07-25:
# Vision
I'm reading components from a graph database, where each component is identified using a UUID. I want to use this UUID as the index for each of the components as I add them into pandapower. 

# Problem
A UUID is essentially a 128-bit integer, and cannot be represented as an int in C (`OverflowError: Python int too large to convert to C long`). UUIDs also have a string representation, but **pandapower seems to deny the use of string indices**. Note that this is not a limitation in `pandas.DataFrame`.

A code snippet to illustrate the problem:
```py
# The uuids are read from a database (where they are the identifiers), not actually generated for the simulation like here.
import uuid
b1_id = str(uuid.uuid4())
b2_id = str(uuid.uuid4())
line_id = str(uuid.uuid4())

import pandapower as pp
net = pp.create_empty_network()
# Using strings as id's here does not cause any errors:
pp.create_bus(net, vn_kv=230, index=b1_id)
pp.create_bus(net, vn_kv=230, index=b2_id)

# And here, when using from_bus and to_bus, the problem arises
# ValueError: could not convert string to float: 'c0d402fe-bca5-4116-a066-17c5dc18a2a1'
pp.create_line_from_parameters(net, index=line_id, from_bus=b1_id, to_bus=b2_id, length_km=1, r_ohm_per_km=1, x_ohm_per_km=1, c_nf_per_km=1, max_i_ka=1)

```

lthurner, 2019-07-26:
This limitation was introduced deliberately, because the internal power flow mapping only works if the indices are integers. The power flow might also be very slow if you use ids as an index, at least for buses.

Instead of mapping the uuids to numbers I would suggest letting pandapower create consecutive indices and keep a mapping of indices to uuids or keep the uuids in a column table to have a way to match uuids to pandapower infices.

bergkvist, 2019-07-26:
@lthurner Okay, I see - so this is related to performance.

Will pandapower perform better with consecutive (small) indices than spread out (large) ones? Of course there is the risk of collisions between the mapped UUIDs, but according to my calculations this risk is low enough that I don't have to care about it.

For 64-bit numbers the chance of one or more collisions of 10,000 random numbers is in the order of 10^(-10). Basically, it is not going to be a problem. (It is around 10^(-8) with 100,000 random numbers).

lthurner, 2019-07-27:
> Will pandapower perform better with consecutive (small) indices than spread out (large) ones? 

Yes. The internal mapping is done on an array, which means an index of 2 is mapped as position 2 in an array. An index of 98123181 would therefore be mapped to position 98123181 of that array. The mapping array would therefore have to become very large, which would block a lot of memory and also not be great for performance. 

We had an implementation once where the lookup was a dictionary, and indices could also have been strings. But iterating over a dictionary is significantly slower than doing a numpy array lookup. Thats why we switched to this solution, even though it puts limits on what indices can look like.

lthurner, 2019-07-30:
I am closing this, since changing the index is not up for discussion currently. Feel free to reopen if you have further questions!

closed 2019-07-30 by lthurner