e2nIEE/pandapower #442: Easy way to merge pp-networks?

thomaswolgast, 2019-07-29:
Hello,

is there an easy way to merge multiple pandapower-networks to a single network? For example, I want to have a network that spans over multiple voltage levels like MV + LV. And I don't want to create it from scratch. Here's how I imagine it:
`merged_net = pp.merge_nets(net1, net2, merge_bus)`

Is something like that possible in pandapower? And if not, do you think that there is need for something like that? It would allow to easily create large multi-level networks from the pre-implemented pandapower networks.

Thank you very much

ascheidl, 2019-07-29:
Hi

see https://github.com/e2nIEE/pandapower/blob/3ec8a5701206f0651b2c3c21cde887a04ff616e2/pandapower/toolbox.py#L1040

cheers,
alex

thomaswolgast, 2019-07-29:
Thank you very much!

So it is possible to merge the networks, but the coupling must be done manually, correct?

ascheidl, 2019-07-29:
yes, you could use a bus-bus-switch for example (has no impediance)

(dont forget to remove the ext_grid of the LV grid if you have one ;))

thomaswolgast, 2019-07-29:
Yes, that is really not much effort. Thank you again.

But I still think automatic coupling would be cool ;) 
(although I assume there are good reasons you did not implement that)

Can be closed!

closed 2019-07-29 by thomaswolgast