e2nIEE/pandapower #444: Faster create functions

ascheidl, 2019-07-30:
The current implemenation of the create_* function is quite slow. Consequently, we implemented for example functions like create_buses to fastly create a bunch of buses at once. However, this means we would need more such functions for most elements loads, sgens and so on.

I tried a different approach. I introduced the option "skip_checks" in create_load which skips the check if an index already exists and also skips the call to _preserve_dtypes. With this modification I can add the intended number of lines in the element table (filled with dummy values) and then use create_load(..., index=x) in a loop to set the correct values. The creation of the additional lines can be done with a convinience function. (To create 1000 loads took 5.5 s in the current and 250ms with the new implementation).

However, from user perspective it might be more complicated. We should discuss if this is a better way.

jdollichon, 2019-07-30:
In a current project i needed to create loads/switches/sgens in bulk for which i implemented analogous to create_buses and could contribute those. As far as i remember the consensus of previous discussions on separate "create-in-bulk" functions was that they would clutter the namespace, which is why we chose to not implement them. I guess there always is a balance to keep between number of functions and number of parameters a user has to be aware of. 

Another option would be to wrap the regular create methods and a "bulk"-version of it in a function and switch behavior depending on whether the input parameter are iterables. This option obviously  comes at the price of some implicity (or reliance on the user reading the docstrings).

lthurner, 2019-07-30:
> I tried a different approach. I introduced the option "skip_checks" in create_load which skips the check if an index already exists and also skips the call to _preserve_dtypes. With this [...]

Interesting, that sounds like a good idea. Did you also make a comparison of create_buses vs. create_bus with the skip_checks parameter?

jdollichon, 2019-08-05:
@ascheidl if you do not have the time to investigate this, could you push the relevant changes to your cloned repo, so i can compare it with bulk-creation?

jdollichon, 2019-08-06:
Down below some timings are provided. Note that the differences become even more severe if the number of elements present before adding new ones is large. Since each call to create_load includes copying the corresponding DataFrame we can easily face runtimes of 30min when adding 20k+ elements.

Original:
`timeit.repeat(setup= 
              "import pandapower as pp\n"
              "import numpy as np\n"
              "net = pp.create_empty_network()\n"
              "bus = pp.create_bus(net, 20)\n"
              "ibp = list(zip(list(range(0, 1000)), np.full((1000,), bus ), np.full((1000,), 0)))",
              stmt=
              "for i, b, p in ibp:\n"
              "    pp.create_load(net, b, p, index=i)", 
              repeat=3, number=1)`

9.613 seconds mean

Accelerated implementation by Alex (identical behavior)
`timeit.repeat(setup= 
              "import pandapower as pp\n"
              "import numpy as np\n"
              "net = pp.create_empty_network()\n"
              "bus = pp.create_bus(net, 20)\n"
              "ibp = list(zip(list(range(0, 1000)), np.full((1000,), bus ), np.full((1000,), 0)))",
              stmt=
              "for i, b, p in ibp:\n"
              "    pp.create_load(net, b, p, index=i)", 
              repeat=3, number=1)`

6.426 seconds mean

Accelerated implementation by Alex (+skip checks)
`
timeit.repeat(setup= 
              "import pandapower as pp\n"
              "import numpy as np\n"
              "net = pp.create_empty_network()\n"
              "bus = pp.create_bus(net, 20)\n"
              "ibp = list(zip(list(range(0, 1000)), np.full((1000,), bus ), np.full((1000,), 0)))",
              stmt=
              "for i, b, p in ibp:\n"
              "    pp.create_load(net, b, p, index=i, skip_checks=True)", 
              repeat=3, number=1)`

1.985 seconds mean

Accelerated implementation by Alex (+skip checks and previous DF extension)
`
timeit.repeat(setup= 
              "import pandapower as pp\n"
              "import numpy as np\n"
              "import pandas as pd\n"
              "net = pp.create_empty_network()\n"
              "bus = pp.create_bus(net, 20)\n"
              "ibp = list(zip(list(range(0, 1000)), np.full((1000,), bus ), np.full((1000,), 0)))\n"
              "dd = pd.DataFrame(index=list(range(0, 10000)), columns=net.load.columns)\n"
              "net['load'] = net['load'].append(dd, sort=False)",
              stmt=
              "for i, b, p in ibp:\n"
              "    pp.create_load(net, b, p, index=i, skip_checks=True)", 
              repeat=3, number=1)`

0.303 seconds mean

Bulk-Creation with separate function
`timeit.repeat(setup= 
              "import pandapower as pp\n"
              "import numpy as np\n"
              "net = pp.create_empty_network()\n"
              "bus = pp.create_bus(net, 20)\n"
              "buses = np.full((1000,), bus)",
              stmt=
              "pp.create_loads(net, buses, 0)",
              repeat=3, number=1)`

0.017 seconds mean

friederikemeier, 2019-08-06:
This was also requested by #466

lthurner, 2019-09-02:
Seems like there is still a significant speedup with the bulk create functions, so we should still keep them, even if the regular create functions can be accelerated. Plan:

- include the speedup of create function that does not change behaviour
- include the flag to omit data type checks to allow speedup of regular create functions
- add bulk create functions for trafo, sgen and load

dlohmeier, 2019-11-11:
I would like to throw in a suggestion that might be a good trade-off, a generic function that adds data to an existing dataframe as bulk (in my example, it can be DatFrame, array or list of lists). This would mean that unexperienced users can still go with the normal create-functions to create examplary networks, whereas especially for converters from other programs or databases this would speed up the network creation. In my example, you can add any kind of element that exists upon creation of an empty network. If you load data from a database that already contains all the columns needed for pandapower, you can just choose the columns you want to include in your pandapower network and give the column names that should overlap with the pandapower column names in order to fit in the data (sorting should however not be an issue).

Here is an example to this:

	def create_element_data(net, element, data, indices=None, columns=None, preserve_dtypes=True,
							use_data_columns=True):
		if element not in net:
			raise UserWarning("The element %s has no table in the given network!" % element)

		# check which columns are already in the table and which are not
		if columns is None and isinstance(data, pd.DataFrame) and use_data_columns:
			columns = data.columns.copy()
		if columns is not None:
			use_data_columns = True
			cols_in_table = [col for col in columns if col in net[element].columns]
			cols_not_in_table = [col for col in columns if col not in net[element].columns]
		else:
			cols_in_table = None
			cols_not_in_table = None

		# create an auxiliary array based on the input data
		if isinstance(data, ndarray) or isinstance(data, list):
			if columns is not None:
				element_data = pd.DataFrame(data, columns=columns)
			else:
				element_data = pd.DataFrame(data)
		elif isinstance(data, pd.DataFrame):
			element_data = data.copy(True)
		else:
			raise UserWarning("The input data type %s is not allowed!" % type(data))

		# define indices
		if indices is None:
			elidx = get_free_id(net[element])
			indices = arange(elidx, elidx + len(element_data), 1)

		# store dtypes for preserving them after adding data
		if preserve_dtypes:
			dtypes = net[element].dtypes

		# extend table with the new indices
		net[element] = net[element].reindex(concatenate([net[element].index.values, indices]))

		# add data to existing table for columns that overlap
		if cols_in_table is not None:
			net[element].loc[indices, cols_in_table] = element_data.loc[:, cols_in_table].values
		else:
			if columns is not None and use_data_columns:
				net[element].loc[indices, columns] = element_data.loc[:, columns].values
			else:
				net[element].loc[indices, :] = element_data.values

		# restore dtypes if possible, if any were altered during data manipulation
		if preserve_dtypes:
			_preserve_dtypes(net[element], dtypes)

		# add columns that were not yet included in the table and infer dtype from given data
		if cols_not_in_table is not None:
			for col in cols_not_in_table:
				net[element][col] = pd.Series(dtype=element_data.dtypes[col])
			net[element].loc[indices, cols_not_in_table] = element_data.loc[:, cols_not_in_table]

		return indices

It is not yet tested for all outcomes and might still require some checks and discussion, but I tested it for some very simple cases and it worked. I think, this would speed up conversion of data immensely and dtypes would still be preserved.

dlohmeier, 2019-11-12:
At least now I added the create_loads function. Sgens and trafos still missing.

bergkvist, 2020-09-01:
## I've created some functions for appending multiple line/switch/load/trafo/trafo3w
Note that they are polymorphic with regards to input size - and also work for appending a single element. They are pure, and doesn't mutate the existing network. The mutation must happen explicitly. The function has no branches in the main body, so it should be easy to cover with tests.

The functions safe_append and size could be tested independently, reducing the need for tests of append_line etc. See the implementation below.

Note that I haven't really needed all the features of the existing create functions; like specifying a custom index when creating elements. Because of that, it is not part of the implementation below. It would probably be relatively easy to add, however.

### Example usage
```py
net.line = append_line(
    net,
    from_bus=np.array([0, 1]), 
    to_bus=np.array([1, 2]), 
    length_km=np.array([0.1, 0.2]), 
    r_ohm_per_km=1,
    x_ohm_per_km=1, 
    c_nf_per_km=1,
    max_i_ka=1
)

display(net.line)
```
[image]

```py
# Appending another line after the existing lines
net.line = append_line(net, from_bus=4, to_bus=5, length_km=1, r_ohm_per_km=2, x_ohm_per_km=2, c_nf_per_km=2, max_i_ka=2)
display(net.line)
```

[image]

---

### Implementation

```py
from pandapower.auxiliary import _preserve_dtypes
from numpy import nan, isnan, arange
import pandas as pd
import numpy as np

def append_line(net, from_bus, to_bus, length_km, r_ohm_per_km, x_ohm_per_km,
                c_nf_per_km, max_i_ka, name=None, type=None, in_service=True,
                df=1., parallel=1, g_us_per_km=0., **kwargs):

    # Validation: Makes sure that the buses are present in the network            
    assert pd.Series(from_bus).isin(net.bus.index).all()
    assert pd.Series(to_bus).isin(net.bus.index).all()
    index = next_free_index(net.line, length=size(from_bus))
    dd = pd.DataFrame(index=index)
        .assign(
            from_bus=from_bus,
            to_bus=to_bus,
            length_km=length_km,
            r_ohm_per_km=r_ohm_per_km,
            x_ohm_per_km=x_ohm_per_km,
            c_nf_per_km=c_nf_per_km,
            max_i_ka=max_i_ka,
            g_us_per_km=g_us_per_km,
            name=name,
            df=df,
            parallel=parallel,
            in_service=in_service,
            std_type=None,
            type=type,
            **kwargs
        )
    return safe_append(net.line, dd)

[... 136 lines cut ...]
            vkr_mv_percent=vkr_mv_percent,
            vkr_lv_percent=vkr_lv_percent,
            pfe_kw=pfe_kw,
            i0_percent=i0_percent,
            shift_mv_degree=shift_mv_degree,
            shift_lv_degree=shift_lv_degree,
            tap_step_percent=tap_step_percent,
            tap_step_degree=tap_step_degree,
            tap_side=tap_side,
            tap_neutral=tap_neutral,
            tap_min=tap_min,
            tap_max=tap_max,
            tap_pos=tap_pos,
            tap_at_star_point=tap_at_star_point,
            name=name,
            in_service=in_service,
            max_loading_percent=max_loading_percent,
            std_type=None,
            **kwargs
        )
        .pipe(lambda dd: dd.assign(tap_pos=dd.tap_pos.fillna(dd.tap_neutral)))
    )
```

</details>

dlohmeier, 2020-09-02:
Thanks for the contribution! How does this complement the create functions defined in PR #842 and merged through #863 ?

SteffenMeinecke, 2021-11-10:
I close this issue since the current implementation of  `create_x()` (e.g. `x="bus"`) and `create_xs()` for all elements with bulk creation need works fine and there is no progress on other implementations.

closed 2021-11-10 by SteffenMeinecke