e2nIEE/pandapower #445: Mistakes in Documentation

thomaswolgast, 2019-07-30:
Hello,

I found some mistakes in the pandapower-documentation: [https://pandapower.readthedocs.io/en/v2.1.0/opf/formulation.html](url)

1. OPF-section: Sometimes still "kilo" is used instead of "mega". For example in the generator constraints table for Q-constraints of the gens and sgens.
2. Same Table as before: The code for load constraints is wrong. "sgen" is used instead of "load".
3. Cost functions: In the examples, "create_polynomial_cost()" is used. But in the tutorial* "create_poly_cost()" is used. I think the first one is wrong.
*Link: [https://github.com/e2nIEE/pandapower/blob/master/tutorials/opf_basic.ipynb](url)
4. The pictures for the cost functions are also outdated (kilo and wrong signs again)
5. Networks-section: Sometimes "import pandapower.networks as pn" is used, sometimes "import pandapower.networks as nw". This is only a small one, but prevents easy copy-pasting of the examples.

I hope that is helpful. Thank you very much for the creation and documentation of pandapower.

Cheers,
Thomas

jdollichon, 2019-07-30:
Those issues arise from the fact, that the link to the documentation referrs to the version that is currently available on PyPi (2.1.0). There have been some changes to the pandapower in the latest version which have not yet been released (see [https://pandapower.readthedocs.io/en/latest/](https://pandapower.readthedocs.io/en/latest/)).

thomaswolgast, 2019-07-30:
Yes, but still there are inconsistencies and mistakes in the documentation which should be fixed. The documentation of 2.1.0 is not consistent with the actual code of 2.1.0.

lthurner, 2019-07-30:
> Yes, but still there are inconsistencies and mistakes in the documentation which should be fixed. The documentation of 2.1.0 is not consistent with the actual code of 2.1.0.

The documentation is included in the github reposirory: https://github.com/e2nIEE/pandapower/tree/develop/doc. If you have found mistakes, feel free to fix them in the respective files and put in a pull request!

thomaswolgast, 2019-07-31:
Created a pull request. Some mistakes were already fixed, some were not. Sorry for the inconvenience, that was my first pull request. I hope I did everything right.

I hope that was helpful.

PS: I could not change the png-files. Therefore, there are still mistakes in the pics (wrong sign + kw instead of mw). Maybe the description text of the pics must be fixed, too.

Cheers,
Thomas

lthurner, 2019-07-31:
> Created a pull request. Some mistakes were already fixed, some were not. Sorry for the inconvenience, that was my first pull request. I hope I did everything right.

Perfect, thanks a lot!

> PS: I could not change the png-files. Therefore, there are still mistakes in the pics (wrong sign + kw instead of mw). Maybe the description text of the pics must be fixed, too.

Right, I will have to check if I find the source files on these...

closed 2019-07-31 by lthurner