e2nIEE/pandapower #448: pp.to_excel saving a grid using geo coords fails.

vogt31337, 2019-07-31:
Hi,

running pandapower develop branch pulled 31.7. 10:00 trying pp.to_excel a grid with bus and line coords (geo coords) fails.

Problem seems to be the function coords_to_df() in io_utils.py, trying to access "coords" in the bus table, which doesn't exist. 

Fix so far is to rearrange the function (starts @ line 50) and pull the part checking for bus up:
```python
def coords_to_df(value, geotype="line"):
    columns = ["x", "y", "coords"] if geotype == "bus" else ["coords"]
    geo = pd.DataFrame(columns=columns, index=value.index)
    if geotype == "bus":
        geo["x"] = value["x"].values
        geo["y"] = value["y"].values
        return geo
    if any(~value.coords.isnull()):
        k = max(len(v) for v in value.coords.values)
        v = numpy.empty((len(value), k * 2))
        v.fill(numpy.nan)
        for i, idx in enumerate(value.index):
            # get coords and convert them to x1, y1, x2, y2...
            coords = value.at[idx, 'coords']
            if coords is None:
                continue
            v[i, :len(coords) * 2] = numpy.array(coords).flatten()
        geo = pd.DataFrame(v, index=value.index)
        geo.columns = ["%s%i" % (w, i) for i in range(k) for w in "xy"]
    return geo
```

BR

lthurner, 2019-08-02:
Usually what we try to do in this case is not adapt the code that it can handle different types of grids, but use the convert_format function to convert grids to the current format. The convert format function is always called when loading a grid. Have you tried to run

> pp.convert_format(net)

before saving the grid to excel? If this doesn't resolve the issue, then we should look into why the convert_format function doesn't add the column.

rbolgaryn, 2019-08-09:
Correct me if I'm wrong, "coords" column is not meant to be in the bus_geodata table in the current pandapower. The standard columns in bus_geodata are x, y and, if it's a GeoDataFrame, geometry.

For saving to Excel, I suggest ignoring the column geometry and just using the columns x and y for buses. 

net.line_geodata has 1 column, coords, and, if it's a GeoDataFrame, geometry (in addition). Here, too, I'd just save the coords to Excel.

SteffenMeinecke, 2021-11-10:
close due to inactivity and no obvious TODOs

closed 2021-11-10 by SteffenMeinecke