e2nIEE/pandapower #456: Why is voltage_depend_loads True by default?

friederikemeier, 2019-08-05:
If I have a net without loads, I get the following error:

`  File "C:\Users\fmeier\pandapower\pandapower\auxiliary.py", line 652, in _init_runpp_options
    if not (np.any(net["load"]["const_z_percent"].values) or np.any(net["load"]["const_i_percent"].values)):
KeyError: 'load'`

I feel like this is unintuitive...

jdollichon, 2019-08-05:
The error you are facing results from the "load" DataFrame not being present in the net. It usually gets created in `https://github.com/e2nIEE/pandapower/blob/5e01b40b7f1018b2606f842caa65cf2568d66ebd/pandapower/create.py#L17` though. The fact that voltage_depend_loads is True does not cause your error. However, i do not know the reasons for the assumption that loads with values set for "const_z_percent" and "const_i_percent" are considered voltage dependend. 

Did you open this issue to discuss the former error or the later assumption? Or did i miss your point entirely?

lthurner, 2019-08-05:
>  Why is voltage_depend_loads True by default?

Because if you have a load and set const_z_percent to say 20%, it would be unintuitive that this setting is ignored by default. The default is to consider all load flow relevant data, which I think is intuitive. The problem here seems to be with the convert_format function? This column should be added if it is not conatined in the grid.

friederikemeier, 2019-11-01:
It was a convert_format problem.

closed 2019-11-01 by friederikemeier