e2nIEE/pandapower #463: Problem installing Pandapower

MagicPatze, 2019-08-06:
Hello comunity, 

when trying to install pandapower with pip i get the following error: 

 ERROR: Could not find a version that satisfies the requirement pandapower (from versions: none)
ERROR: No matching distribution found for pandapower

When I check my installed python version, it says that Python 3.7.3 is installed 

I hope you can help me

lthurner, 2019-08-06:
Are you maybe behind a firewall?

MagicPatze, 2019-08-06:
First of all, thanks for the fast reply!

That could be actually, since it tells me that "an existing connection has been closed by the remotehost". 
For which program do i have to create an exception on the firewall, though? 
I created exceptions for _conda and python, it still does not work.

lthurner, 2019-08-06:
This is a general issue about how to use pip with a firewall. I am closing this issue as it is not related to pandapower, there should be plenty of information on this to find on stackoverflow or other places, please have a look there.

closed 2019-08-06 by lthurner