e2nIEE/pandapower #468: Strange behaviour on Linear Powerflow currents

Gseguelg, 2019-08-07:
Hi. During linear powerflow, the following simple system gives the astonishing same result for current (and therefore loading percent) for this line. 
```
import pandapower as pp

reactance = 0.05
net = pp.create_empty_network(sn_mva=1, add_stdtypes=False, f_hz=50)
pp.create_bus(net, vn_kv=1, name='Bus 0')
pp.create_bus(net, vn_kv=1, name='Bus 1')
pp.create_ext_grid(net, 0, vm_pu=1, vang_deg=0)
pp.create_line_from_parameters(net, name='Line 1-2', from_bus=0, to_bus=1, length_km=1, r_ohm_per_km=0, x_ohm_per_km=reactance, c_nf_per_km=0,  max_i_ka=0.5)
pp.create_load(net, 1, p_mw=1)
pp.rundcpp(net)
# pp.runpp(net)
print(net.res_line.loc[0, :])

reactance *= 5
net.line.at[0, 'x_ohm_per_km'] = reactance
pp.rundcpp(net)
# pp.runpp(net)
print(net.res_line.loc[0, :])
```

Note that series reactance is 5 time greater, but current [kA] won't change. This is wrong according to the power transfer approximation equation. Also, I've noticed it only happens for linear powerflow.

Cheers.

wangzhenassd, 2019-08-07:
Hi,
in DC Power Flow  P is proportional to the angle's difference and B. Voltage magnitude is considered 1 p.u. and no losses are considered. If you keep the load power and B constant, no current change should be observed in DC power flow but only the angle's difference.

Gseguelg, 2019-08-07:
@wangzhenassd exactly.

The thing is that here B is a single value (at bus 2), and should be the inverse of that changed susceptance. Therefore B changes and so do the voltage angles but not the current, not correct.

Other way to see it is to think on term of ohm's law, where current magnitud depends on voltage drop between the buses (in magnitud) and series impedance magnitud (here reactance). So current should be different as angles and/or reactance changes.

wangzhenassd, 2019-08-07:
In DC load flow all buses have voltage of 1 p. u., it is just the angle difference over X and all P flow on branches will be calculated.  if you want to consider voltage drop,  then you have to use normal AC load flow.

Gseguelg, 2019-08-08:
I might have a translation issue here. What I meant with "voltage droop" is the equivalent voltage phasor resulting of subtracting calculated voltages at buses connected. The problem is that linear powerflow assumptions don't give easy information from this point of view. Also in my example, B changes but also does the voltage angles in the same proportion. Therefore the effect is called out (This can be proven by equations).

Before I realise that, I was confusing my self with this one line example and a multiple lines one. In the simple example current/power demanded by the load has to flow entirely through the only existing line. If at least another line exist, current/power should flow (without power losses) according to lines reactances. So in a multiline example, current through each line will vary if it's reactance is changed but total power has to be the same.

closed 2019-08-09 by Gseguelg