e2nIEE/pandapower #472: copying a network

Gseguelg, 2019-08-09:
Hi. Is there a direct function/method to make an independent copy of a pandapower network?

I've seen that the pandapowerNet class inherits from predefined ADict class, which inherits from a common python dictionary. The last one has a method copy() to return an independent copy on the dictionary. Yet when you call net.copy() the whole dictionary will be returned.

If you agree and there is no copy method implemented, I can make a pull request to add a new method on pandapowerNet class for returning an independent network, rather than the whole dictionary/ADict.

wangzhenassd, 2019-08-11:
Hi,
net2 = copy.deepcopy(net) will do the trick

Gseguelg, 2019-08-12:
Nice one. I had in mind creating a new instance of the network class. This should be a faster than importing a new module, but i think copy gets imported anyway and the time difference can be minimal.

`net2 = pp.pandapowerNet(net)` or `net2 = pp.pandapowerNet(net.copy())`

Can the copy method from a pandapower network be overwritten? Or is it preferable to have a different name?

Gseguelg, 2019-08-12:
I made some raw timing:
```
>>> timeit.timeit("net2=copy.deepcopy(net)", setup="import copy;import pandapower.networks as ppn;net=ppn.case9()", number=1000)
8.128615273002652
```

```
>>> timeit.timeit("net2 = pp.pandapowerNet(net)", setup="import pandapower as pp;import pandapower.networks as ppn;net=ppn.case9()", number=1000000)
4.6094630810002855
```

There might be a significant difference. Please correct me if I'm wrong.

wangzhenassd, 2019-08-13:
Tried the following, the function won't give a new object for the internal dataframes

    import pandapower.networks as nw
    import pandapower as pp
    net = nw.simple_four_bus_system()
    net2 = pp.pandapowerNet(net)

    print(id(net.load), id(net2.load))
    >>>242217200 242217200

Gseguelg, 2019-08-15:
Yes, it's true. `copy.deepcopy()` will do the best for now. Thank you for the commit, I just haven't had the time nor your agreement to do it. I see you also incorporated the option to initialize a network as a copy from another. Looks good.

What about defining the method as an attribute? I think it will be faster to use when required.

wangzhenassd, 2019-08-15:
> Yes, it's true. `copy.deepcopy()` will do the best for now. Thank you for the commit, I just haven't had the time nor your agreement to do it. I see you also incorporated the option to initialize a [...]

What do you main exactly by defining the method as an attribute?

Gseguelg, 2019-08-15:
> What do you main exactly by defining the method as an attribute?

Define deepcopy method as a _getter_, while it requires no arguments (at least yet).

> PandapowerNet has actually a copy() method because of the inheritance from "dict", but this returns only a shallow copy. So I added a deepcopy method which will do the same of "copy.deepcopy".

Yes, saw the commit. Thanks.

wangzhenassd, 2019-08-16:
> Define deepcopy method as a _getter_, while it requires no arguments (at least yet).
>
>
> Yes, saw the commit. Thanks.

OK, then I think the PR should meet the request. Since copy is a generally a function in Python, I don't see a point why deepcopy should be defined as property.

FlorianShepherd, 2019-08-16:
fixed by f0f4f45 
Thanks @wangzhenassd and @Gseguelg

closed 2019-08-16 by FlorianShepherd