e2nIEE/pandapower #480: How to incorporate start-up & shutdown costs?

Misbahrani, 2019-08-19:
Hello,
I want to implement the start-up and shutdown costs of the generators. AS the operational cost estimation using both polynomial and piece-wise linear functions is easily done in pandapower, I was wondering to expand the cost estimation,and want to incorporate startup costs, so at OPF the cheapest generators is turned ON first. Plus i want to incorporate ramp limits and start-up time. How to do that in pandapower framework?
Best regards,
Misbah Rani

lthurner, 2019-09-02:
Hi, good question, this would indeed be very helpful feature. The reason this is so tricky is that this adds discrete optimization variables to your problem which makes it harder to solve. I don't think this is possible with the PYPOWER OPF that is used in pandapower by default. The best possibility is probably to do this is by going through the PowerModels interface, which allows a pretty modular definition of the optimization problem. Have you checked if this is possible in PowerModels?

Misbahrani, 2019-09-02:
Thanks for replying. I don't know about powermodels.

lthurner, 2019-09-02:
You can find a documentation about how to use pandapower with powermodels here: https://pandapower.readthedocs.io/en/v2.1.0/opf/powermodels.html

You can first check out the capabilities of PowerModels here: https://lanl-ansi.github.io/PowerModels.jl/stable/ or if you don't find any information on this open an issue on the powermodels repo: https://github.com/lanl-ansi/PowerModels.jl

Misbahrani, 2019-09-02:
Okay thanks a lot for helping me, I really appreciate it. I will check them
out.

closed 2020-05-18 by lthurner