e2nIEE/pandapower #489: Meaning of calculate_voltage_angles in runpp

BasecampMaster, 2019-09-02:
I don't understand the exact meaning of calculate_voltage_angles.
Does calculate_voltage_angles = False lead to a simplification for the power flow computation?
I did a test on bigger meshed MV grid (roughly 400 nodes) without any significant errors for voltages and voltages angles in the cases calculate_voltage_angles = True and calculate_voltage_angles = False.

However, running runpp with a highly meshed grid with about 1400 nodes leads to singularity problems (affected function: numpy.linalg.solve) with calculate_voltage_angles  = True. The simulation only terminates successfully with calculate_voltage_angles = False. In the latter case, I witness significant differences for voltages and voltage angles compared to other powerflow computation tools.

What is the exact meaning of calculate_voltage_angles in runpp?
Is pandapower designed to work with bigger grids, too?

Amonara, 2019-09-02:
We also had a problem with calculate_voltage_angles = True. The solution was to add some PV-nodes `pp.create_gen` to our network. There are more then 1500 nodes in the network and it works.

FlorianShepherd, 2019-09-02:
The transformer phase shift values are not being considered if calculate_voltage_angles = False

For meshed grids the value of calculate_voltage_angles defaults to True. This is automatically checked based on the voltage level (above 70kV the value of calculate_voltage_angles == True). See:
https://pandapower.readthedocs.io/en/v2.0.1/powerflow/ac.html?highlight=calculate_voltage_angles#power-flow 

pandapower is designed to work bis "bigger" grids. We tested up to 100k buses. However, you can get numerical issues depending on your initial voltage setpoint. Have you tried to use a DC init before the AC power flow?

lthurner, 2019-09-02:
> The transformer phase shift values are not being considered if calculate_voltage_angles = False

Also the voltage angle setpoints of ext_grids and generators.

You should always use calculate_voltage_angles=True if you have multiple ext_grids or generators that are electrically connected in the grid. If you have numerical problems when calculating voltage angles, you should have a look at the initialization as suggested by Florian.

BasecampMaster, 2019-09-03:
Thank you for all the answers. The DC init for the initial voltage setpoints doesn't lead to a success unfortunately.

Do I understand correctly that calculate_voltage_angles = True has to be set for meshed grids?

FlorianShepherd, 2019-09-03:
> Do I understand correctly that calculate_voltage_angles = True has to be set for meshed grids?
yes

closed 2019-10-04 by lthurner