e2nIEE/pandapower #492: SQLite import/export bus_geodata

jurasofish, 2019-09-09:
It seems that pandapower does not save the bus_geodata table properly when using the pp.to_sqlite function.

``` python
import pandapower as pp
net = pp.create_empty_network()
bus1 = pp.create_bus(net, vn_kv=110)
bus2 = pp.create_bus(net, vn_kv=110)
pp.create_ext_grid(net, bus1)
pp.create_line(net, bus1, bus2, length_km=10,
               std_type="N2XS(FL)2Y 1x300 RM/35 64/110 kV")
pp.create_load(net, bus2, p_mw=2, q_mvar=4)

pp.to_sqlite(net, './net.sqlite')

net2 = pp.from_sqlite('./net.sqlite')
```
```
Traceback (most recent call last):
  File "C:/Users/mjurasovic/.PyCharmCE2019.1/config/scratches/sqlite_pp_error.py", line 12, in <module>
    net2 = pp.from_sqlite('./net.sqlite')
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandapower\file_io.py", line 411, in from_sqlite
    net = from_sql(con)
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandapower\file_io.py", line 404, in from_sql
    net = from_dict_of_dfs(dodfs)
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandapower\io_utils.py", line 161, in from_dict_of_dfs
    net = df_to_coords(net, item, table)
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandapower\io_utils.py", line 139, in df_to_coords
    net[item].loc[:, ['x', 'y']] = table.loc[:, ['x', 'y']]
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandas\core\indexing.py", line 1404, in __getitem__
    return self._getitem_tuple(key)
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandas\core\indexing.py", line 808, in _getitem_tuple
    retval = getattr(retval, self.name)._getitem_axis(key, axis=i)
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandas\core\indexing.py", line 1814, in _getitem_axis
    return self._getitem_iterable(key, axis=axis)
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandas\core\indexing.py", line 1119, in _getitem_iterable
    keyarr, indexer = self._get_listlike_indexer(key, axis, raise_missing=False)
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandas\core\indexing.py", line 1078, in _get_listlike_indexer
    keyarr, indexer, o._get_axis_number(axis), raise_missing=raise_missing
  File "D:\jp\programs\miniconda3\envs\xxxx\lib\site-packages\pandas\core\indexing.py", line 1163, in _validate_read_indexer
    key=key, axis=self.obj._get_axis_name(axis)
KeyError: "None of [Index(['x', 'y'], dtype='object')] are in the [columns]"

```

The bus_geodata table has only an index column, but df_to_coords seems to assume the other columns are present ([cursor.description](https://docs.python.org/3/library/sqlite3.html#sqlite3.Cursor.description) returns a tuple of tuples where the first elements are the column names)

``` python
import sqlite3
with sqlite3.connect('./net.sqlite') as con:
    c = con.cursor()
    c.execute('select * from bus_geodata').fetchall()
    print(c.description, c.fetchall())
```

```
(('index', None, None, None, None, None, None),) []
```

ascheidl, 2020-01-16:
is this still an issue ? your code is working for me in the current version

jurasofish, 2020-01-20:
Thanks for the bump, just tested on 2.2.0 and works fine

closed 2020-01-20 by jurasofish