e2nIEE/pandapower #493: Poly cost

thomaswolgast, 2019-09-13:
Hello,
I have two issues about the OPF and the poly_cost function:

First issue, I want to perform an OPF to reduce active power losses using the following costs:
```
for actuator in ['sgen', 'ext_grid']:
    for idx in net[actuator].index:
        pp.create_poly_cost(net, idx, actuator, cp1_eur_per_mw=p_price)
for idx in net.load.index:
    pp.create_poly_cost(net, idx, 'load', cp1_eur_per_mw=-p_price)
```
But when I look at `net.res_cost` after OPF, the (negative) costs for the loads are not considered at all, which results in far too high total costs. Why is that so? Because my loads are not controllable? I understand that they are not considered for optimization for that reason, but in my opinion the results should be stored with the correct costs, because otherwise that is a possible source for mistakes, if the resulting costs are used for further work.

That leads to my second issue.  I perform multiple optimizations where I change the reactive power price `cq2_eur_per_mvar2` continuously. But when I looked at `net.poly_cost`, I saw that the prices don't get updated, but instead there are now multiple costs for a single unit. How are these poly_cost values used internally? Are all these costs added together? Or is only the most recent value used for optimization? Again, I think that is very error-prone and should get changed.

Thank you very much

lthurner, 2019-09-17:
>  Why is that so? Because my loads are not controllable? 

Yes

> in my opinion the results should be stored with the correct costs, because otherwise that is a possible source for mistakes, if the resulting costs are used for further work.

The costs are a result of the OPF calculation, so the costs only consider the variables considered in the OPF. However I agree with you that it might lead to misunderstandings, its worth thinking about changing this convention.

> That leads to my second issue. I perform multiple optimizations where I change the reactive power price cq2_eur_per_mvar2 continuously. But when I looked at net.poly_cost, I saw that the prices [...]

Can you please make a minimal code example, its not quite clear to me what you are doing.

thomaswolgast, 2019-09-18:
Thank you! 

Minimal example:
```
net = pn.example_simple()
pp.create_poly_cost(net, 0, 'sgen', cp1_eur_per_mw=1, cq1_eur_per_mvar=1)
pp.create_poly_cost(net, 0, 'sgen', cp1_eur_per_mw=2, cq1_eur_per_mvar=2)
print(net.poly_cost)
```
In my opinion, the outcome should be:
```
   element    et  cp0_eur  cp1_eur_per_mw  cp2_eur_per_mw2  cq0_eur  cq1_eur_per_mvar  cq2_eur_per_mvar2
1      0.0  sgen      0.0             2.0              0.0      0.0               2.0                0.0
```

Instead, the actual outcome is:
```
   element    et  cp0_eur  cp1_eur_per_mw  cp2_eur_per_mw2  cq0_eur  cq1_eur_per_mvar  cq2_eur_per_mvar2
0      0.0  sgen      0.0             1.0              0.0      0.0               1.0                0.0
1      0.0  sgen      0.0             2.0              0.0      0.0               2.0                0.0
```

So I defined two cost functions for the exact same generation unit. But for me it is unclear now which of these values is used for the OPF-calculation. As far as I see it, the second line should overwrite the first one and not get appended, to ensure clarity. Or did I understand something wrong here? They can't be used both, right?

SteffenMeinecke, 2021-11-10:
Getting the answer with a simple trial: The last entry is used.
To avoid such implicit and arbitrary behaviour, I am implementing checks to create functions and runopp().

commit 9cda2e21 (2021-11-16): add checks to create functions and runopp(); close #493
closed 2021-11-16 by SteffenMeinecke