e2nIEE/pandapower #505: Different comparison methods of pandapower elements (dataframes)

dlohmeier, 2019-09-27:
In pandapower there are two different ways of how dataframes are compared. On the one hand there is an auxiliary function `dataframes_equal` in the toolbox, which is also used in the auxiiliary function `nets_equal`. On the other hand in the JSONSerializableClass dataframe and series objects are compared using `assert_series_equal` and `assert_frame_equal` from `pandas.util.testing`. I think the latter is generally better suited for two reasons:

1. `assert_frame_equal` seems to be better at casting frames with dtypes object into the correct form. This is especially important for the case that a frame contains mixed integer and NaN values in which case the function `dataframes_equal` in pandapower does not recognize this behaviour and fails to filter out those NaNs.
2. `dataframes_equal` sorts indices / columns inplace which is normally not very problematic, but will definitely be an unexpected behaviour for the user. So after calling `pp.nets_equal`, the nets might look different than before. This could of course simply be avoided by using copies.

dlohmeier, 2019-10-01:
Furthermore, there is the function `assert_net_equal` in `pandapower.test.toolbox`. This function differs from `pandapower.toolbox.nets_equal`, as it only compares certain tables, which actually seems a bit implicit to me. There is also ony way of making this function throw an error (KeyError), although it is just meant to return True/False, which is when one of the tables is only present in one of the nets to compare. I would suggest that it makes sense to try and integrate allthe different comparison methods and throw out those that are not needed any more.

SteffenMeinecke, 2019-11-06:
>     2. `dataframes_equal` sorts indices / columns inplace which is normally not very problematic, but will definitely be an unexpected behaviour for the user. So after calling `pp.nets_equal`, the nets might look different than before. This could of course simply be avoided by using copies.

This is not an opinion, just a hint in case you didn't see it: the auxiiliary function `nets_equal` has the flag `ignore_index_order` to not resort index.

SteffenMeinecke, 2021-11-11:
This issue should consider https://github.com/e2nIEE/pandapower/issues/505

closed 2021-12-02 by SteffenMeinecke