e2nIEE/pandapower #511: optimal power flow generator limits

FlorianShepherd, 2019-09-30:
The generator / ext_grid OPF behaviour is imo not clear. For example there is the opf/test_basic.py test_convert_format() test with a generator specified like this:

pp.create_gen(net, 1, p_mw=0.1, controllable=True, min_p_mw=0.005, max_p_mw=0.15,
                  max_q_mvar=0.05, min_q_mvar=-0.005)

Here p_mw is set, but not the voltage. The voltage limits (vmin, vmax) are taken from the corresponding bus instead. I think it would be better to tighten the voltage of the generator to vm_pu, which must be mandatory like in the power flow. If someone wants a P, Q controlled generator (which I think is the intended behaviour in the case above), I'd expect an sgen with  P, Q limits defined and "controllable=True". 

This gen behaviour is especially a little bit confusing / inconsistent at the moment when considering the following cases:

- A grid with multiple ext_grids in the same area. All but one of these ext_grids (slacks) are internally converted to gens and vm_pu is fixed for the gens in small boundaries. (fixed here 62dafaf94ce52efb3bf4840bdf92423e35f673d3)
- When only gens are in the grid and one is specified as "slack", then the voltage is *not* fixed for each generator to vm_pu of the gen. (As in the example above). 

@friederikemeier @lthurner opinions?

FlorianShepherd, 2019-09-30:
We agreed on the following behaviour:

- gens get additional V limits (vmin, vmax)
- the parameter controllable == True (default)
- if controllable == False -> vm and p_mw are enforced to their current value
- if controllable == True -> vmin < v < vmax is taken from gen if set. if it is not set it is taken from bus. p_mw must be set for gen (otherwise warning) in this case
- if the bus and the gen have V limits, the intersection is taken as limits. 
- if vm_pu value is outside of this limit an error is raised.

I'll implement this

FlorianShepherd, 2019-09-30:
While implementing I came across the following case:

```
create_gen(net, 0, p_mw=0., vm_pu=1., max_p_mw=1., min_p_mw=-1., min_q_mvar=-1, max_q_mvar=1., slack=True) -> controllable == True, slack == True -> what to do?

pp.create_gen(net, 1, p_mw=0.02, vm_pu=1.01, max_p_mw=1., min_p_mw=-1., min_q_mvar=-1, max_q_mvar=1., controllable=False) # controllable == False -> vm_pu forced to 1.01

pp.create_gen(net, 3, p_mw=0.01, vm_pu=1.02, max_p_mw=1., min_p_mw=-1.,
                       min_q_mvar=-1, max_q_mvar=1.)  # controllable == True -> vm_pu between bus voltages
```

The first generator is a slack bus with vm_pu set. However, controllable == True (meaning "consider the bus max_vm_pu / min_vm_pu values"). The OPF converges to 1.008577 p.u. in my test, which seems fine. The question is what is expected here? Is slack == True more important than controllable==True and vm_pu should be enforced?
@friederikemeier @lthurner opinions?

FlorianShepherd, 2019-10-01:
fixed most of the tests:
a49425bba2edee8ba17bae6941aeda7aaab1f918

except the DC line tests. These seem to be highly unstable. 
https://github.com/FlorianShepherd/pandapower/blob/develop/pandapower/test/opf/test_dcline.py

If vm_pu for ext_grids is set with a delta here in line 111 / 112:
pandapower/build_gen.py 

the DC line tests fail because the OPF doesn't converge. I tried to find a delta which works (see the get_delta_try_except() in test_dcline.py), but it didn't succeed...

friederikemeier, 2019-10-09:
@rbolgaryn check out the changes!

closed 2019-10-09 by FlorianShepherd