e2nIEE/pandapower #516: [Feature] create standard library by request

Gseguelg, 2019-10-03:
As seen from #510, I got the idea of creating standard type library by request at network creation time. Maybe the implemented way wasn't the best. Here the reasons why I think it's a good idea.

1. There is no need to have a standard library when they are not used (e.g., when creating lines from parameters).
2. To drop standard library after default creation doesn't make sense from an efficiency point of view.
3. Default behaviour might stay the same, i.e., standard library is created to avoid conflicts and ease of use for beginners.
4. Studies might require use of standard library for some elements (like transformers), but for others might not (like lines).

Therefore, the idea is to modify `create_empty_network`, `add_basic_std_types` and `from_dict_of_dfs` to make it more explicit then commit n°2 from #510.
What do you think?

lthurner, 2019-10-04:
> 1. There is no need to have a standard library when they are not used (e.g., when creating lines from parameters).

There isn't, but the question is if there is any downside to having them.

> 2. To drop standard library after default creation doesn't make sense from an efficiency point of view.

How is it actually inefficient? Are you talking about execution time, memory or disk space? Because for all of them I would assume the impact of adding the standard types would actually be minimal. This sounds very much like premature optimization to me.  A simple timing of this doesn't seem like adding the standard types makes any difference 

[image]

> Studies might require use of standard library for some elements (like transformers), but for others might not (like lines).

Yes, but always adding all standard types also solves this problem. Can you provide any quantifiable example where adding standard types that are not used is actually a problem with regard to efficiency?

lthurner, 2019-11-01:
Closed for inactivity, if there is need for further discussion on this, please reopen

closed 2019-11-01 by lthurner