e2nIEE/pandapower #526: pandapower data structure - which columns to add by default and which on the fly?

lthurner, 2019-10-15:
With growing analysis capabilities of pandapower, we have added many additional parameters for the different elements. Some of them are added when creating an empty network, others are not considered as part of the core data structure, and we add them on the fly. 

All power flow parameters are mandatory and added by default. For other analysis, certain columns have to be added:
- OPF: gen.max_p_mw, gen.min_p_mw, etc.
- symmetric short circuit: ext_grid.s_sc_max_mva, gen.xdss_pu etc.
- unbalanced short circuit or power flow: line.c0_nf_per_km, trafo.vector_group etc. (see #523)

Until now our philosphy has been that these columns do not exist in the data frames, but are created if they are specified in the element creation. So creating a gen with minimum active power for OPF:

```pp.create_gen(net, min_p_mw=10)```

adds the column is added to the data frame. The downside of that is that the user does not see which columns need to be specified, because they are not in the data frame until they are actually specified. My suggestion would be to switch to a system where the columns that should be added are specified at empty_network_creation:

```
pp.create_empty_network(add_opf_parameters=True)
```

or 

```
net = pp.create_empty_network()
pp.add_opf_parameters(net)
```

FlorianShepherd, 2019-10-15:
I'd prefer the second option, since I just stumbled upon the following case:

1.  load an existing network without elements with additional parameters (like max_p_mw in the OPF)
2. add an element with the additional parameters set. Example: `pp.create_bus(..., max_vm_pu=1.1, min_vm_pu = .9`

The bus DataFrame has now one entry with `vm` values set. The other buses do not have this setpoint / any value for min / max vm_pu.

Therefore, it would be great to have something like:
 ```
pp.add_opf_parameters(net, default_values={"bus": {"min_vm_pu":0.9, max_vm_pu": 1.1}}, overwrite=False)
```

which adds all columns necessary for the OPF and then sets the specified default values for each element if no value is defined. (Or it overwrites the set values when `overwrite==True`)

FlorianShepherd, 2019-10-15:
Maybe also a toolbox function which drops the non-default values / calculation specific values would be great. So one can reduce the size of the grid.

omar49san, 2020-01-23:
Dear Pandapower Users, 
i'm using this great opensource tool in my loadflow and short circuit calculation, every thing work good for now, however i am facing some difficulties in calculating earth to ground fault current (1ph short circuit current). I know that it needs zero sequence parameters of elements constituting the network (cables, transformers,etc..). In the IEC60909 standards, i can find some zero sequence values for the majority of the electrical elements except the  zero sequence parameters of transformer such **Vkr0, Vk0, sio_hv_partial, mag0_percent**, i had seen your example for 1ph short circuit calculation using pandapower and  where you had defined zero sequence parameters of transformer. So my question is :
Did you use some specific formulas to calculate this zero sequence parameters or you had picked it from datasheet transformer ? In case you had used a formulas, could you share any reference or biblioraphic ressource that can help to find it?
I think you in advance for your answer and for all your efforts in making this opensource tool very useful and easy to use.

shankhoghosh, 2020-01-28:
Hi @omar49san ,

The transformer modeling is done using a T model transformer,

I would suggest 
https://www.aeso.ca/assets/linkfiles/4040.002-Rev02-Transformer-Modelling-Guide.pdf
for reference,

As for the additional parameters

**vk0 & vkr0** are generally the same and mag0_percent depends on the type of transformer.
core-type(3-limb) :  3 -10
Shell-type transformer (4/5-limb) : 10-100

**si0_partial_hv** is the distribution of SC impedance towards hv side. So, 0.9 value means most of the impedance is towards the HV side.

We took the sample values from a commercial software standard type, but if you can find something with zero sequence parameters of distribution transformers, you are welcome to share the source.

vk0 and vkr0 are basically for leakage impedance (copper losses) and mag0_percent is a ratio between Iron and copper losses and iron losses for zero-sequence (Zmag0 / Zsc0 ).

omar49san, 2020-01-31:
Hi @shankhoghosh 
I'm very glad to see your answer, i do really appreciate the time you had spent to answer my questions and for the valuable informations you had feed me with. 
I find the transformer guide reference you had shared with me very rich and useful, i have a better understanding now about the zero sequence component.
I agree that it is very hard to obtain accurate values of zero sequence parameters for any transformer with a specific power and winding connection, unless the manufacturers of transformers agreed to give you their zero sequence parameters obtained from the tests they had made on the different transformer sizes(which seems a very hard task because no manufacturer will choose to share freely his data whatever the reason is ). And of course, i will have the pleasure to share with you any of this values parameter for distribution transformer as soon as i get it.
Concerning si0_partail_hv ,I had read some where that it can be assumed that the SC impedance is distributed equally in the two sides of transformer which mean 50% of Sc impedance in the primary side and the other 50% impedance in the secondary side. Do you find this assumption acceptable ?

SteffenMeinecke, 2021-11-10:
Conclusion so far:

- Questions of @omar49san don't fit to this issues title. If still relevant, please open a new issue.

- Optional but valuable TODO: toolbox function which drops the non-default values / calculation specific values

heckstrahler, 2026-11-09:
This is now handled via the network structure introduced with pandera. In the pandera schemata the optional columns are tagged accordingly  and when an empty network is created only the loadflow required columns are created. It is also possible to create a new empty network with optional columns via parameter depending on what you need.

closed 2026-11-09 by heckstrahler