e2nIEE/pandapower #551: Question in _init_ppc function

bbadetcauk, 2019-11-05:
Hello, 

In ppc_conversions.py there is a statement to run a dc power flow if there are phase shifting transformers in the model. 

```
# do dc power flow for phase shifting transformers
    if np.any(net.trafo.shift_degree):
        vm_backup = ppci["bus"][:, 7].copy()
        ppci["bus"][:, [2, 3]] = 0.
        ppci = _run_dc_pf(ppci)
        ppci["bus"][:, 7] = vm_backup
```

I think that the line `ppci["bus"][:, [2, 3]] = 0.` is setting p and q injections (loads, sgens) at all busses to zero in the ppci for the remainder of the calculation. 

I am using the State Estimation tool which I think is looking for existing p and q injections from elements (loads, sgens) in the model (when using zero-injection='auto') before making a decision whether each bus should be considered a 'zero-injection bus' or otherwise.

By assigning these columns to zero in the ppci (as a result of phase shifting transformers==True) it is changing the number of zero injection busses considered during state estimation. Is this the desired behaviour?

Please forgive me if I have not expressed my question correctly - this is my first post on Github! Thanks

Ben

jhmenke, 2019-11-06:
It looks like a bug. @wangzhenassd  can you comment on this?

wangzhenassd, 2019-11-06:
Bug fixed in PR #548

bbadetcauk, 2019-11-06:
Many Thanks! Great work that you are doing with Pandapower, will try to support as a learn more.

FlorianShepherd, 2019-11-19:
@wangzhenassd Can this be closed?

wangzhenassd, 2019-11-20:
> @wangzhenassd Can this be closed?

Yep, we can merge the PR today

closed 2019-11-20 by wangzhenassd

jhmenke, 2019-11-20:
it is now merged