e2nIEE/pandapower #552: LoadflowNotConverged: PP Diagnostic Tool intepretation

kvrdz, 2019-11-05:
Hallo everyone,

I'm rather new to Powerflow estimations and Pandapwer in general. I'm trying to simulate a simple distribution network, on running the power flow calculations Load flow non convergence error.
On running the Diagnostic tool, I get the following : 

1) overload:
overload found: Power flow converges with load scaled down to 0.1 percent.

2) wrong_switch_configuration:
Power flow still does not converge with all switches closed.

Is 1) Overload error due to the Transfomer capacity being equal to the load?

Transformer
`pp.create_transformer_from_parameters(net, name='trafo_1',hv_bus=0, lv_bus=1,
**sn_mva=0.315** ,vn_hv_kv=10, vn_lv_kv=0.4,vk_percent=4, vkr_percent=1.325, 
pfe_kw=0.95,i0_percent=0.2375)`
 
Load
` #create load  Keine reactive loads nur ressistiv d.h q_mvar = 0
pp.create_load(net,bus=2,p_mw=0.315)`

2)Power flow still does not converge with all switches closed. : could you please tell me how I must interpret this?

Thanks in advance!

lthurner, 2019-11-08:
Sometimes the power flow fails to invalid data, sometimes just because the load or generation is so large, that the problem is infeasible. To check this, the diagnostic function scales the load and runs a power flow with the scaled load. The diagnostic function tells you the power flow converges with a scaled down load, so the reason it doesn't converge is that the load is to high. You should check if the values you assumed are realistic.

There are some odd switch configurations that sometimes lead to non-convergence. To check this, all switches are closed to see if the power flow converges with closed switches. In your case it doesn't, so the switch configuration is not your problem.

kvrdz, 2019-11-10:
Thanks, i figured it out, it was to do with my load values-

closed 2019-11-10 by kvrdz