e2nIEE/pandapower #572: Measurement elements

bbadetcauk, 2019-12-05:
Hello,

In documentation section > State Estimation > Defining Measurements > Types of Measurements states the measurement should be given in Amps:

_“i” for electrical current measurements at a line (in A)_

Further down in the 'create_measurement' function description, states value should be in kA:

_value (float) - Measurement value. Units are “MW” for P, “MVar” for Q, “p.u.” for V, “kA” for I. Generation is a positive bus power injection, consumption negative_

Please can you confirm  whether A or kA is required. 

Also, with regards to power flow direction for real power measurements - if we have a line between bus 0 (from) and bus 1 (to), power flow is from bus 0 (from) to bus 1 (to), and the measurement is placed on bus 1 (to) side, can you confirm that the real power value should be negative (since it is being consumed by adjacent bus)? And visa versa if the measurement was at bus 0, the real power flow would be positive?

wangzhenassd, 2019-12-05:
Hi,
it'a typo mistake since pandapower just updates the estimation module for phasor measurements. It should be kA. I will make a PR on that. 

The flow direction corresponds the pandapower results. In this case P line at bus 0 should correspond P_from_mw. You can verify this with loadflow results. For bus, injections are positive, which is opposite to pandapower res_bus.

wangzhenassd, 2020-01-17:
Closed due to inactivity

closed 2020-01-17 by wangzhenassd