e2nIEE/pandapower #580: OPF verbose bug

camiloromers, 2019-12-10:
There's a little bug when your print OPF results, even in the develop version. The number of transformers are not correct. It seems that it takes the same number of lines. Here's the code I am using:

```
import pandapower.networks as ppnw
net = ppnw.case14()
net.poly_cost.loc[0, 'cp1_eur_per_mw'] = 100
net.gen.loc[[1,2,3], 'p_mw'] = [20, 30, 10]
net.gen.loc[[1, 3], 'controllable'] = False
pp.rundcopp(net, verbose=True)
print (net.res_gen)
```

And the results in the summary report (how many?) are incorrect. You might check as well in all tutorials you have and you will see a similar behavior.

FlorianShepherd, 2019-12-18:
@camiloromers are you using the pp.opf_task(net) function or what do you mean by summary report?

FlorianShepherd, 2019-12-18:
This report is misleading. 

The "report" function is a legacy from pypower / matpower. The printed results reflect the gens in the internal ppc format, which is the one used by the OPF. The report is not referring to the pandapower elements. 

A quick fix would be to add a notice. However, a report function with the pandapower elements would be nice.

camiloromers, 2019-12-18:
In every OPF report, the number of transformer are always the same the number of buses. In the current case, should be 5 and not 20. I compared with others grid and the same bug is present.

[image]

FlorianShepherd, 2019-12-18:
So the values are also wrong for the ppc?

closed 2020-01-28 by FlorianShepherd