e2nIEE/pandapower #584: pandapower installation failing in some rare cases

dlohmeier, 2019-12-12:
The installation via pip is a little tricky when it comes to C-libraries. As pointed out at https://www.pandapower.org/start/ , in case a user does not use Anaconda to work with pandapower (e.g. when just using miniconda), some packages have to be pre-installed with their respective C-libraries before installing pandapower via pip.

I just tried this and installed just the named packages via miniconda and pandapower via pip. Then I got an error upon importing pandapower as in the `io_utils` package, shapely is imported which requires geos which again includes some C-libraries. The error should be handled, however in case a C-library is not found, this does not throw an ImportError, but an OSError. That's why the exception in `io_utils` is not handled properly. So, I think there are two options:
1. add an OSError to the exception (as it doesn't seem mandatory to have shapely installed for pandapower)
2. include geos and maybe shapely to the required package list in https://www.pandapower.org/start/ .

I think that should be easily fixable for the next release.

lthurner, 2020-01-02:
I think we should just go with option 1., because option 2. might make the installation of pandapower unnecessarily difficult for users on windows, where shapely and geos are not installable through pip.

Can you put in a PR?

dlohmeier, 2020-01-06:
I added this and merged the PR.

closed 2020-01-06 by dlohmeier

DEEPS15121, 2025-04-22:
Jupyter notebook launch error:
Traceback (most recent call last):
File "C:\Users\deepa\anaconda3\lib\site-packages\notebook\traittypes.py", line 235, in _resolve_classes
klass = self._resolve_string(klass)
File "C:\Users\deepa\anaconda3\lib\site-packages\traitlets\traitlets.py", line 2015, in _resolve_string
return import_item(string)
File "C:\Users\deepa\anaconda3\lib\site-packages\traitlets\utils\importstring.py", line 33, in import_item
module = __import__(package, fromlist=[obj])
ModuleNotFoundError: No module named 'jupyter_server.contents'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "C:\Users\deepa\anaconda3\Scripts\jupyter-notebook-script.py", line 10, in 
sys.exit(main())
File "C:\Users\deepa\anaconda3\lib\site-packages\jupyter_core\application.py", line 269, in launch_instance
return super().launch_instance(argv=argv, **kwargs)
File "C:\Users\deepa\anaconda3\lib\site-packages\traitlets\config\application.py", line 1073, in launch_instance
app = cls.instance(**kwargs)
File "C:\Users\deepa\anaconda3\lib\site-packages\traitlets\config\configurable.py", line 583, in instance
inst = cls(*args, **kwargs)
File "C:\Users\deepa\anaconda3\lib\site-packages\traitlets\traitlets.py", line 1292, in __new__
inst.setup_instance(*args, **kwargs)
File "C:\Users\deepa\anaconda3\lib\site-packages\traitlets\traitlets.py", line 1335, in setup_instance
super(HasTraits, self).setup_instance(*args, **kwargs)
File "C:\Users\deepa\anaconda3\lib\site-packages\traitlets\traitlets.py", line 1311, in setup_instance
init(self)
File "C:\Users\deepa\anaconda3\lib\site-packages\notebook\traittypes.py", line 226, in instance_init
self._resolve_classes()
File "C:\Users\deepa\anaconda3\lib\site-packages\notebook\traittypes.py", line 238, in _resolve_classes
warn(f"{klass} is not importable. Is it installed?", ImportWarning)
TypeError: warn() missing 1 required keyword-only argument: 'stacklevel'